rerun-io/rerun #2537: Exclude testing archetypes/datatypes/components from release artifacts

teh-cmc, 2023-06-27:
We have a lot of objects that are only used for testing purposes and that generate _a lot_ of useless code as far as end users are concerned. Make sure they don't pollute e.g. the wasm artifact!

teh-cmc, 2023-07-07:
Becoming more and more urgent as the affix fuzzing tests grow exponentially with all the union stuff.

teh-cmc, 2023-07-25:
Turned this into a tracking for all 3 supported languages.

emilk, 2023-08-08:
Can this be solved once and for all for all three languages by ignoring the test .fbs files in release builds?

Wumpf, 2023-08-08:
fbs files are only looked at at codegen time. Having different codegen for release & debug implies having two different libraries for debug and release, so that doesn't make sense.
Each language implementation needs to put testing files outside of its core sdk library or exclude it somehow from its debug build (this is still cluttering though for anyone browsing the sdk, so separate place on a per-language place is best).

closed 2023-09-20 by teh-cmc