rerun-io/rerun #2551: Irritating error on spawn with an instance already running

Wumpf, 2023-06-28:
Currently initializing with `spawn=True` will try to spawn a Rerun viewer but fail with this error message if one is already running on the same port:
```
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/Users/andreas/dev/rerun-io/rerun/rerun_py/rerun_sdk/rerun/__main__.py", line 16, in <module>
    main()
  File "/Users/andreas/dev/rerun-io/rerun/rerun_py/rerun_sdk/rerun/__main__.py", line 12, in main
    exit(bindings.main(sys.argv))
         ^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: Failed to bind TCP address "0.0.0.0:9876". Another Rerun instance is probably running. Address already in use (os error 48)
```
.. and then silently connect to the existing one.

We should instead have two clearly defined options:
* `force_spawn`: Fail spawning if a viewer is already running on that port
* `connect_or_spawn`: Connect if a viewer is already running, otherwise spawn a new one

I'm suggesting to deprecate `spawn` and have it point to `connect_or_spawn` instead.

emilk, 2023-09-23:
`force_spawn` could be called `spawn_new`, and automatically find a free port

emilk, 2023-09-25:
Let's improve the error message for 0.9

closed 2023-09-28 by jleibs