rerun-io/rerun #2556: The viewer remains sometimes empty when running from IDE

abey79, 2023-06-28:
This is a weird one: when running a simple script from an IDE, the viewer sometimes remains empty. I've experienced this repeated with PyCharm, and something similar has recently been reported by a Discord user using VSCode ([discord convo](https://discord.com/channels/1062300748202921994/1075873257124810852/1123250356449710132)).

I can reproduce with a certain reliability using this:
```python
import time
import rerun as rr

rr.init("test", spawn=True)
#time.sleep(2)
rr.log_points("points", [(0, 0), (2, 2), (2, 2.5), (2.5, 2), (3, 4)], radii=0.5)
```

Uncommenting the `sleep()` call appears to reduce the rate of occurence.

I've *never* seen that when launching from the CLI, same for the discord user.

This resemble #2124, but is probably different, as the `sleep` call is before the logging call, rather than at the end. It's as if the viewer takes too long to start and the logged messages are dropped.

Wumpf, 2023-06-29:
I'm seeing this issue more prominently when I stream to a file instead of connecting to a viewer - the first call (typically view_coordinates -.-) may be lost then.
Critical issue imho, nothing worse than getting flaky behavior

jleibs, 2023-07-24:
We should investigate this before the release, but timebox the investigation.