rerun-io/rerun #2576: Crash when changing space view visiblity

Wumpf, 2023-06-30:
Don't know the exact repro steps (wasn't able to provoke it again) but I managed to crash arkit scenes by changing space view visibilities in blueprint panel (or maybe I clicked on remove?)

```
thread 'ThreadId(1)' panicked at 'index out of bounds: the len is 1 but the index is 1', 3fff2b0/src/container/grid.rs:194

   8: core::panicking::panic_fmt
             at core/src/panicking.rs:64:14
   9: core::panicking::panic_bounds_check
             at core/src/panicking.rs:159:5
  10: <usize as core::slice::index::SliceIndex<[T]>>::index
             at core/src/slice/index.rs:260:10
      core::slice::index::<impl core::ops::index::Index<I> for [T]>::index
             at core/src/slice/index.rs:18:9
      <alloc::vec::Vec<T,A> as core::ops::index::Index<I>>::index
             at alloc/src/vec/mod.rs:2703:9
      egui_tiles::container::grid::Grid::layout
             at 3fff2b0/src/container/grid.rs:194:78
  11: egui_tiles::container::Container::layout
             at 3fff2b0/src/container/mod.rs:220:38
  12: egui_tiles::tiles::Tiles<Pane>::layout_tile
             at 3fff2b0/src/tiles.rs:362:13
  13: egui_tiles::tree::Tree<Pane>::ui
             at 3fff2b0/src/tree.rs:194:13
  14: re_viewport::viewport::Viewport::viewport_ui::{{closure}}
             at re_viewport/src/viewport.rs:557:13
      core::ops::function::FnOnce::call_once{{vtable.shim}}
             at core/src/ops/function.rs:250:5
  15: <alloc::boxed::Box<F,A> as core::ops::function::FnOnce<Args>>::call_once
             at alloc/src/boxed.rs:1987:9
  16: egui::ui::Ui::scope_dyn
             at egui-0.22.0/src/ui.rs:1732:19
  17: egui::ui::Ui::scope
             at egui-0.22.0/src/ui.rs:1720:9
      re_viewport::viewport::Viewport::viewport_ui
             at re_viewport/src/viewport.rs:547:9
  18: <alloc::boxed::Box<F,A> as core::ops::function::FnOnce<Args>>::call_once
             at alloc/src/boxed.rs:1987:9
  19: <alloc::boxed::Box<F,A> as core::ops::function::FnOnce<Args>>::call_once
             at alloc/src/boxed.rs:1987:9
      egui::containers::frame::Frame::show_dyn
             at egui-0.22.0/src/containers/frame.rs:225:19
  20: egui::containers::frame::Frame::show
             at egui-0.22.0/src/containers/frame.rs:216:9
      egui::containers::panel::CentralPanel::show_inside_dyn
             at egui-0.22.0/src/containers/panel.rs:1022:9
      egui::containers::panel::CentralPanel::show_inside
             at egui-0.22.0/src/containers/panel.rs:1007:9
      re_viewer::app_state::AppState::show::{{closure}}
             at re_viewer/src/app_state.rs:163:17
      core::ops::function::FnOnce::call_once{{vtable.shim}}
             at core/src/ops/function.rs:250:5
  21: <alloc::boxed::Box<F,A> as core::ops::function::FnOnce<Args>>::call_once
             at alloc/src/boxed.rs:1987:9
  22: <alloc::boxed::Box<F,A> as core::ops::function::FnOnce<Args>>::call_once
             at alloc/src/boxed.rs:1987:9
      egui::containers::frame::Frame::show_dyn
             at egui-0.22.0/src/containers/frame.rs:225:19
  23: egui::containers::frame::Frame::show
             at egui-0.22.0/src/containers/frame.rs:216:9
      egui::containers::panel::CentralPanel::show_inside_dyn
             at egui-0.22.0/src/containers/panel.rs:1022:9
  24: egui::containers::panel::CentralPanel::show_inside
             at egui-0.22.0/src/containers/panel.rs:1007:9
      re_viewer::app_state::AppState::show
             at re_viewer/src/app_state.rs:141:9
  25: re_viewer::app::App::ui::{{closure}}
             at re_viewer/src/app.rs:530:29
      core::ops::function::FnOnce::call_once{{vtable.shim}}
             at core/src/ops/function.rs:250:5
  26: <alloc::boxed::Box<F,A> as core::ops::function::FnOnce<Args>>::call_once
             at alloc/src/boxed.rs:1987:9
  27: <alloc::boxed::Box<F,A> as core::ops::function::FnOnce<Args>>::call_once
             at alloc/src/boxed.rs:1987:9
      egui::containers::frame::Frame::show_dyn
             at egui-0.22.0/src/containers/frame.rs:225:19
  28: egui::containers::frame::Frame::show
             at egui-0.22.0/src/containers/frame.rs:216:9
      egui::containers::panel::CentralPanel::show_inside_dyn
             at egui-0.22.0/src/containers/panel.rs:1022:9
  29: egui::containers::panel::CentralPanel::show_dyn
             at egui-0.22.0/src/containers/panel.rs:1050:30
  30: egui::containers::panel::CentralPanel::show
             at egui-0.22.0/src/containers/panel.rs:1034:9
      re_viewer::app::App::ui
             at re_viewer/src/app.rs:494:9
      <re_viewer::app::App as eframe::epi::App>::update
             at re_viewer/src/app.rs:864:9
  31: eframe::native::epi_integration::EpiIntegration::update::{{closure}}
             at eframe-0.22.0/src/native/epi_integration.rs:500:13
      egui::context::Context::run
             at egui-0.22.0/src/context.rs:374:9
  32: eframe::native::epi_integration::EpiIntegration::update
             at eframe-0.22.0/src/native/epi_integration.rs:498:27
  33: <eframe::native::run::wgpu_integration::WgpuWinitApp as eframe::native::run::WinitApp>::run_ui_and_paint
             at eframe-0.22.0/src/native/run.rs:1281:21
  34: eframe::native::run::run_and_return::{{closure}}
             at eframe-0.22.0/src/native/run.rs:152:17
  35: <winit::platform_impl::platform::app_state::EventLoopHandler<T> as winit::platform_impl::platform::app_state::EventHandler>::handle_nonuser_event::{{closure}}
      winit::platform_impl::platform::app_state::EventLoopHandler<T>::with_callback
             at winit-0.28.6/src/platform_impl/macos/app_state.rs:70:13
      <winit::platform_impl::platform::app_state::EventLoopHandler<T> as winit::platform_impl::platform::app_state::EventHandler>::handle_nonuser_event
             at winit-0.28.6/src/platform_impl/macos/app_state.rs:91:9
  36: winit::platform_impl::platform::app_state::Handler::handle_nonuser_event
             at winit-0.28.6/src/platform_impl/macos/app_state.rs:199:21
  37: winit::platform_impl::platform::app_state::AppState::cleared
             at winit-0.28.6/src/platform_impl/macos/app_state.rs:388:13
  38: winit::platform_impl::platform::observer::control_flow_end_handler::{{closure}}
             at winit-0.28.6/src/platform_impl/macos/observer.rs:79:21
      winit::platform_impl::platform::observer::control_flow_handler::{{closure}}
             at winit-0.28.6/src/platform_impl/macos/observer.rs:41:9
      std::panicking::try::do_call
             at std/src/panicking.rs:487:40
      std::panicking::try
             at std/src/panicking.rs:451:19
      std::panic::catch_unwind
             at std/src/panic.rs:140:14
      winit::platform_impl::platform::event_loop::stop_app_on_panic
             at winit-0.28.6/src/platform_impl/macos/event_loop.rs:245:11
      winit::platform_impl::platform::observer::control_flow_handler
             at winit-0.28.6/src/platform_impl/macos/observer.rs:39:5
      winit::platform_impl::platform::observer::control_flow_end_handler
             at winit-0.28.6/src/platform_impl/macos/observer.rs:74:9
  39: <unknown>
  40: <unknown>
  41: <unknown>
  42: <unknown>
  43: <unknown>
  44: <unknown>
  45: <unknown>
  46: <unknown>
  47: <unknown>
  48: <unknown>
  49: winit::platform_impl::platform::event_loop::EventLoop<T>::run_return::{{closure}}
             at winit-0.28.6/src/platform_impl/macos/event_loop.rs:220:22
      objc2::rc::autorelease::autoreleasepool
             at objc2-0.3.0-beta.3.patch-leaks.3/src/rc/autorelease.rs:313:5
      winit::platform_impl::platform::event_loop::EventLoop<T>::run_return
             at winit-0.28.6/src/platform_impl/macos/event_loop.rs:211:25
  50: <winit::event_loop::EventLoop<T> as winit::platform::run_return::EventLoopExtRunReturn>::run_return
             at winit-0.28.6/src/platform/run_return.rs:51:9
      eframe::native::run::run_and_return
             at eframe-0.22.0/src/native/run.rs:132:5
  51: eframe::native::run::wgpu_integration::run_wgpu::{{closure}}
             at eframe-0.22.0/src/native/run.rs:1478:17
      eframe::native::run::with_event_loop::{{closure}}
             at eframe-0.22.0/src/native/run.rs:116:9
      std::thread::local::LocalKey<T>::try_with
             at std/src/thread/local.rs:446:16
      std::thread::local::LocalKey<T>::with
             at std/src/thread/local.rs:422:9
  52: eframe::native::run::with_event_loop
             at eframe-0.22.0/src/native/run.rs:109:5
      eframe::native::run::wgpu_integration::run_wgpu
             at eframe-0.22.0/src/native/run.rs:1475:13
```

emilk, 2023-08-25:
Looks like it needs a fix in `egui_tiles`

emilk, 2023-09-12:
I fail to reproduce.

I've cleaned up the code in https://github.com/rerun-io/egui_tiles/pull/26
I cannot see how it could possibly crash now.

Perhaps this only happened on some old `egui_tiles` version.

closed 2023-09-12 by emilk