rerun-io/rerun #2579: Implement blueprint storage on web-view

jleibs, 2023-06-30:
https://github.com/rerun-io/rerun/pull/2578 introduces the ability to persist blueprint data for desktop app.

We need to provide similar implementations for the web-view where we don't have access to file-system storage.

jleibs, 2023-07-04:
Would probably be best to do this by extending eframe: https://github.com/emilk/egui/issues/3134

Wumpf, 2024-07-09:
@jleibs We're doing this by now, right?

jleibs, 2024-07-09:
> @jleibs We're doing this by now, right?

Not to my knowledge. But also I'm not convinced it's necessary. 

I kind of like that there's not a magical hidden browser cache where we persist blueprint state. Either you explicitly provide blueprint via url param (or manual open/drag-and-drop),or you get the default behavior.