rerun-io/rerun #2591: Re-enable codegen on Windows

teh-cmc, 2023-07-03:
Codegen (specifically, `re_types_builder/build.rs` & `re_types/build.rs`) has been disabled on Windows in <TODO:LINK> because it was somehow computing different sha256 sums (??) and thus breaking the CI.

This isn't too bad of an issue as it only affects contributors on windows that are trying to modify IDL definitions but should nonetheless be fixed sooner than later.

jleibs, 2023-10-05:
The sha-mismatch is likely another case where we are likely hitting line-ending inconsistencies introduced by git.

Also, when generating the files on windows we need to take care to normalize our path-separators. Running `cargo codegen` on windows now modifies the path comment in every file.

emilk, 2024-02-28:
`re_types/build.rs` is gone, and the hashing is completely client-side (no hashes checked in), so I think this issue no longer applies.

closed 2024-02-28 by emilk