rerun-io/rerun #2627: Data missing when using Rust SDK

emilk, 2023-07-07:
Minimal reproduce:

```rust
use rerun::{
    components::{ColorRGBA, Point3D},
    time::{TimeType, Timeline},
    MsgSender, RecordingStreamBuilder,
};

fn main() -> anyhow::Result<()> {
    rerun::external::re_log::setup_native_logging();

    let (rec_stream, storage) = RecordingStreamBuilder::new("images").memory()?;
    let timeline = Timeline::new("keyframe", TimeType::Sequence);

    let pt = Point3D {
        x: 1.0,
        y: 1.0,
        z: 1.0,
    };

    MsgSender::new("world/pts")
        .with_time(timeline, 1)
        .with_component(&[pt])?
        .with_splat(ColorRGBA::from_rgb(161, 34, 212))?
        .send(&rec_stream)?;

    rerun::native_viewer::show(storage.take())?;

    Ok(())
}
```

Results in this:

[image]

Seems like data is lost somehow. Some flushing that is not being done?

emilk, 2023-07-07:
Adding a call to `rec_stream.flush_blocking();` before `storage.take();` solves it.

Seems like we should at least warn when data is being silently lost, or better yet: make `take()` block until data is flushed.

teh-cmc, 2023-07-07:
Yeah, I think the most user-friendly solution here would be to do the same thing we do in the jupyter notebook internals in Python, namely: internally keep track of the `RecordingStream` associated with this `MemorySinkStorage`, and flush on behalf of the user where it makes sense.

It's a bit blurry how all of this works though, I'll have a quick look to see what's possible.

PR #2632 (2023-07-07): Automagic flush when `take()`ing a `MemorySinkStorage`
Fixes #2627 
Should supersede #2630 

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun.io](https://demo.rerun.io/pr/2632) (if applicable)

- [PR Build Summary](https://build.rerun.io/pr/2632)
- [Docs preview](https://rerun.io/preview/pr%3Acmc%2Fautoflushmagix/docs)
- [Examples preview](htt [... cut]
commit 9ccaabfa (2023-07-07): Automagic flush when `take()`ing a `MemorySinkStorage` (#2632)

Fixes #2627 
Should supersede #2630 

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun.io](https://demo.rerun.io/pr/2632) (if
applicable)

- [PR Build Summary](https://build.rerun.io/pr/2632)
- [Docs
preview](https://rerun.io/preview/pr%3Acmc%2Fautoflushmagix/docs)
- [Examples
pre [... cut]
closed 2023-07-07 by teh-cmc