rerun-io/rerun #2635: Speed up CI

emilk, 2023-07-07:
The CI is so slow these days. I can't take it.

abey79, 2023-07-11:
I just noticed that, is it known?

[image]

jprochazk, 2023-07-21:
After a few PRs:
- https://github.com/rerun-io/rerun/pull/2740
- https://github.com/rerun-io/rerun/pull/2753

I'm happy to report that we have shaved off anywhere from 8 to 12 minutes from the "main" path of the PR CI runs. 

- before: https://github.com/rerun-io/rerun/actions/runs/5554804192
- after: https://github.com/rerun-io/rerun/actions/runs/5613492308

What this means is `build web + build wheel -> build demo app -> upload web demo` takes about 8-12 minutes, where previously it took about 17-20 minutes. (Side note: rendering done by GHA here is not great, jobs overlap in strange ways, you have to hover a job to see the actual dependency path between some subset of the jobs)

There's still a lot of room for improvement. Most of it will come from https://github.com/rerun-io/rerun/issues/2762.

I believe we can squeeze much more performance out of GHA with some better job sequencing and/or (ab)using Docker.

Wumpf, 2023-07-21:
Lot of work done this cycle. Let's create a new tracking issue for remaining items @jprochazk

closed 2023-07-21 by Wumpf

jprochazk, 2023-07-21:
https://github.com/rerun-io/rerun/issues/2776