rerun-io/rerun #2647: Code-gen for C++

emilk, 2023-07-10:
To generate the C++ object hierarchy there are a few things to figure out.

One of them is `.hpp` vs .cpp files. This is probably quite easy.

## Sum types
How do we express sum types in C++?

Consider a simple example (in Rust):

```rust
pub enum Rotation3D {
    Quaternion(Quaternion),
    AxisAngle(RotationAxisAngle),
}

```

### Tagged unions
For Plain Old Data (POD) a tagged union is very efficient:

```cpp
enum Rotation3DKind {
    Quaternion,
    AxisAngle,
};

union Rotation3DData {
    Quaternion quaternion;
    AxisAngle axis_angle;
};

struct Rotation3D {
    kind: Rotation3DKind,
    data: Rotation3DData,

    // implicit conversions:
    Rotation3D(Quaternion);
    Rotation3D(AxisAngle);
};
```

However, this becomes quite error prone if the contained data is not plain-old-data, but has allocations.

It is still doable if we make the union and all the fields private. We would need to take care to overload (or make private):
* The copy constructor
* The move constructor
* Copy assignment
* Move assignment
* Destructor

See https://en.cppreference.com/w/cpp/language/rule_of_three.

In other words, it would have to be something like:

```cpp
namespace {  // private
  enum Rotation3DKind {
      Quaternion,
      AxisAngle,
  };
  
  union Rotation3DData {
      Quaternion quaternion;
      AxisAngle axis_angle;
  };
} // namespace

namespace rr {
    class Rotation3D {
      private:
        _kind: Rotation3DKind,
        _data: Rotation3DData,
  
      public:
        // Implicit constructors (maybe should be explicit?):
        Rotation3D(Quaternion&& quaternion) {
            _kind = Rotation3DKind::Quaterion;
            _data.quaternion = quaternion;
        }

        Rotation3D(AxisAngle&& axis_angle);
        
        ~Rotation3D {
             switch self.kind {
                 Rotation3DKind::Quaterion:
                     self.data.quaterion.~Quaterion();
                     break;

                 Rotation3DKind::AxisAngle:
                     self.data.axis_angle.~AxisAngle();
                     break;
             }
        }
    }
}
```

### Inheritance
Another approach is to use inheritance. This adds a lot of allocation and indirections though, which will be quite costly.

abey79, 2023-07-10:
FWIW, some notes from the (Python) front:
- For Python, the core tool I'm using is `typing.Union[arm_type1, arm_type2, ...]`, which obviously is of no use for C++.
- Using inheritance would require (near) systematic wrapping of the arm's type in some type, since you can't have a `Vec3D` or a `float` directly inherit from `Scale3D`. For Python, this was a big no go in terms of the impact on the API. 
- `Angle { Degrees(float), Radians(float) }` was... interesting. I basically introduce a union tag field `kind` any time 2 or more arms share the same type, as (dynamic) type info is no longer sufficient to discriminate.

Question: would it be feasible to back-port (and improve for our purpose) `std::variant` or does it depend on core language features?

emilk, 2023-07-10:
I don't think we want to back-port `std::variant` even if we could. For one, it uses the the type-id of the thing it holds, i.e. cannot distinguish between float degrees and radians unless degrees and radians are their own new-types. `std::variant` is also overall very unergonomic, so we would still want to generate wrappers around it.

Wumpf, 2023-08-07:
With #2916 all the basics are in place, we "just" need to extend to more complex types.
Further tracking happening in 
* #2919

closed 2023-08-07 by Wumpf