rerun-io/rerun #2651: Frustum lines flicker in and out when tracking 3D camera

Wumpf, 2023-07-10:
See post by @javrtg in https://github.com/rerun-io/rerun/issues/2638#issuecomment-1628627942

Wumpf, 2023-07-10:
This seems to be happening due to some frame delay. We can probably fix this by piping through the information to the camera scene part if it is tracked and then simply not render the frustum lines - they are annoying in any case while tracking a camera

nikolausWest, 2023-07-10:
[image]

emilk, 2023-07-14:
Was this solved by https://github.com/rerun-io/rerun/pull/2667?
* https://github.com/rerun-io/rerun/pull/2667

Wumpf, 2023-07-18:
PR linked to the wrong issue. this one here is fixed.

closed 2023-07-18 by Wumpf