rerun-io/rerun #2661: Better handle non-array archetype fields in Python API V2

abey79, 2023-07-10:
Currently, non-array fields of archetypes are handled like arrays, and noting keeps the user to actually provide an array of data. Example: `Points2D.draw_order`. This needs to be improved so the API is more explicit about it and provides safeguards.

nikolausWest, 2023-07-10:
Not sure if that is what you meant but I'd much prefer if we let users log both scalars and arrays rather than requiring them to wrap scalars with arrays.

abey79, 2023-07-10:
@nikolausWest wrapping in array is never required. Any field that accept an array will also accept a single instance, as one would expect. Some fields (such as `Points2D.draw_order`) should *only* accept a scalar (and refuse an array). This is currently not the case, and that's what this issue is about. Its probably not a huge deal though, I expect the viewer might just disregard all draw_order values but the first.

abey79, 2023-11-06:
Solved with the archetype `__init__` overrides

closed 2023-11-06 by abey79