rerun-io/rerun #2682: Put more thoughts into entity references

teh-cmc, 2023-07-12:
Entity references is something we've talked about a bunch of times in the past, and even did a proof-of-concept at some point (#1487).

The general idea is that sometimes you really want a component to just be a reference to an existing component in another entity (or maybe even in the same one..??!).

Consider e.g. the case of a textured `Mesh3D`: you might want to add an `AlbedoTexture` component directly to that mesh (let's say it's e.g. a `Tensor` under the hood) or maybe you just want to point to an existing image-like thing that's already available in another entity.

Something akin to the following:
```
table Mesh {
  vertex_positions: [[float: 3]];
  vertex_colors: [uint] (nullable);
  vertex_normals: [[float: 3]] (nullable);

  indices: [u32] (nullable);

  albedo_texture: AlbedoTexture (nullable);
}

table AlbedoTexture {
  texture: _AlbedoTexture;
  vertex_uvs: [[float: 2]];
}

enum _AlbedoTexture {
  entity_path: EntityPath,
  tensor: Tensor,
}
``` 

No real plans yet, just something to keep in mind.

teh-cmc, 2023-08-02:
This actually is becoming high priority as we start implementing blueprint features.

Even the most basic definition of a `SpaceView`, by definition, has to define some kind of `Query` object that is just a bunch of entity path references.

teh-cmc, 2024-05-16:
We now have two different tools that could do this kind of thing:
- On the blueprint side, a reference could be implemented as part of the query DSL, and the reference is "resolved" ahead of time by the query planner.
- On the promise side, a reference could be implemented using e.g. a store URI (`store://a/b/c:Color`), and the reference is resolved by the promise resolver.

gavrelina, 2024-10-28:
Adding a relevant use case, requested by a user: 

Context: SLAM
A user wants to be able to refer to the same keyframe:
- when these keyframes get created (he logs a new entity for keyframe-id, in his example it's `map/keyframe/1/...`). they are like 'source of truth'
- and then on each input frame (which they call `query_image`, they need to find N closest keyframe, labeled `kf0`, `kf1`, `kf2`, etc.)

Right now to get the UI to sort of do what he wants, they fully re-log the whole keyframe structure at those locations. For example maybe at time X `last_query/kf1` is a full copy of `map/keyframe/7`, but then later at time Y  `last_query/kf1` might instead be a copy of `map/keyframe/37`.

he is looking for something like a link/alias/reference 
``` rr.log('last_query/kf0', Reference('map/keyframe/12')) ```

This would let the viewer do two things:
- For creating the view of `last_query/kf0` it would just use the data from the referred entity.
- If he selects/overs over the referenced entity (`map/keyframe/3`), then we would also highlight (maybe in a different color) all of the aliases that point to that location.  So maybe `last_query/kf9` would also light up.

cc @jleibs