rerun-io/rerun #2691: New API to request userspace-callbacks from interaction events in visualizer

jleibs, 2023-07-13:
A recurring request from users is a mechanism for user-interactions such as clicking to generate a callback in user-space code.

Some examples:
 - Click on a point and receive the timepoint + entity-path + instance-id
 - Show an arbitrary list in a UI element and receive an event from the selection

TODO:
 - More details on roadmap for this feature

jaidevshriram, 2023-08-07:
Hey! I'd love this feature too. I went through the mentioned PR but it looks like it's meant for Rust and not Python. If there is a pythonic way to create custom space views, I'd appreciate any help. Thanks!

Wumpf, 2023-08-08:
@jaidevshriram yes, the linked PR is sadly Rust only and this is a very heavy-handed way of extending the viewer. This ticket is indeed all about making smaller adjustments from all languages in a more data-driven way :)
Unfortunately, this is a bit further down the line; we're first going to look into more basic control of the viewer layout from code (if you see us talk about blueprint in tickets and on discord that's that)

karanchahal, 2024-04-10:
Hello,

If there is a way to expose a javascrip library that will enable usage of rerun specific things with custom UI components, that would significantly help in using this tool for custom tools on top of the renderer/viewer etc.

alik-git, 2024-04-20:
Just echoing the need for this feature on my end as well. I'm making a 3D mapping system for robotics that allows you to search for objects in the scene via natural language, and this would help a lot. 

See: https://github.com/concept-graphs/concept-graphs

My specific need is: Having something like a text input box with a labelled "search" button, that when filled out and pressed would trigger a callback to our code that performs the search and opens a new view with the desired object, or recolors all the objects in the scene by how relevant they are to the search query.

This would really help people migrate more easily from other options to rerun. Thanks!! Rerun is awesome.

Erol444, 2024-06-19:
+1 on user inputs, like buttons/text inputs/sliders (perhaps API could be similar to streamlit one?). OpenCV provides sliders, so I use opencv just for that, and then rerun for visualziation.

bokorn-bdaii, 2024-06-27:
+1 this with python interfaces for mouse over and selection in images (expanding graph nodes) as well as keyboard callbacks.

Famok, 2024-08-15:
I would love that feature for annotating my datasets

Lukasdoe, 2025-01-25:
+1 for a C++ callback API for buttons / user input

zalo, 2025-04-09:
+1 for the callback API with extremely basic buttons, sliders, and text boxes.

As such, I think I'm going to have to roll my own visualization framework, since Rerun doesn't make sense right now for anything requiring user input or with multiple servers feeding one client.

Wumpf, 2025-04-14:
>  multiple servers feeding one client

We have that already! :)
Two processes (no matter where) can feed the same Viewer by specifying the same appliation & recording id

Wumpf, 2025-04-14:
General update: 0.23.0 will introduce some basic callbacks for the npm package, jupyter notebooks and self-hosted Rust builds of the Viewer.

Wumpf, 2026-05-04:
The in-process callbacks have been extended a bit but nothing in the way of extending the UI of the viewer. For that you still need to either write rust code https://rerun.io/docs/howto/visualization/extend-ui or embedd it in a website that adds some ui (there, the existing callbacks come in handy, see https://ref.rerun.io/docs/js/stable/web-viewer/)