rerun-io/rerun #2692: Proposal: standardize on singular type descriptions and "Batch" suffixes

jleibs, 2023-07-13:
Something I've noticed, at least on the python side, is that our flexible handling of single-vs-batched APIs within archetype constructor signatures introduces a lot of room for error and ambiguity.  This is further exacerbated by handling of splats which allow for the singular type variant in places otherwise expecting batched inputs.

Additionally, I currently find the subtlety of `Point2d` (the datatype) vs `Points2d` the archetype to be non-obvious and a very easy source of typos/confusion.

Proposal:
- Always describe archetypes, components, and datatypes in terms of native singular type.
- Add an attribute to the IDL schema to indicate whether types support the batch APIs.
- For any batched type, also generate an explicit `<>Batch` type to complement the singular type.
- Maybe: Require splats to be annotated explicitly when logging.

From an API perspective this would mean we would have:
```
rr.log("point", rr.Point2d([1,2])

rr.log("points", rr.Point2dBatch([[1,2],[3,4]])

rr.log("points", rr.Point2dBatch([[1,2],[3,4]], colors=rr.Splat(0xff0000))
```

Archetypes like image and transform, which don't support batching, would not even have a `Batch` variant.

If `Point2dBatch` is too verbose, we could still add additional type-aliases for common cases  `Points2d = Point2dBatch`

Incorrectly shaped inputs, such as the following, would generate errors.
```
rr.log("point", rr.Point2d([[1,2]])
rr.log("points", rr.Point2dBatch([1,2])
```

emilk, 2023-07-14:
In this case I wonder if we shouldn't rename these types anyway. `Points2D` is the archetype, with a `position: Pos2` component, which is implemented with the `Vec2` datatype. It also has the `radius: Radius` component, implemented with the `Scalar` datatype. etc

jleibs, 2023-07-14:
Totally agreed on the Pos rename, I was holding off on that at this point until after we have query support for the new types in the viewer though.

(The usage of 3D is unfortunate... Point2DBatch both looks uglier than Point2dBatch and also causes the spell-checker to complain while the lower-case version is accepted.)

jleibs, 2023-07-14:
How about using a sub-type as in `Point2D.Batch`? It both avoids the 2DBatch CamelCase annoyance, and also makes it explicitly clear that it's a related/derived type-signature as opposed to a parallel native type.

jleibs, 2023-07-14:
Closing as rejected.

closed 2023-07-14 by jleibs