rerun-io/rerun #2727: Remove hover effect in checkboxes

martenbjork, 2023-07-18:


abey79, 2023-07-24:
Here is the current situation, which applies to all controls:
```rust
{
    // Expand hovered and active button frames:
    egui_style.visuals.widgets.hovered.expansion = 2.0;
    egui_style.visuals.widgets.active.expansion = 2.0;
    egui_style.visuals.widgets.open.expansion = 2.0;
}
```
Setting `hovered.expansion` to 0 removes the scaling on hover.

**Questions**:
- Should the scaling be removed on click as well (`active.expension = 0.0`? I would say this is not necessary as there is already some click feedback with thickening of the white check mark.
- How about `open.expension`? It refers buttons dropdown menu, see 2nd video.
- Since this is a rather global setting, do we expect undesirable side-effects (e.g. other types of widgets would _should_ scale on hover and/or activation)?

This is with both hovered and active set to 0.0:

https://github.com/rerun-io/rerun/assets/49431240/9ffef4a1-30c5-4801-a8e8-f2478e6e3ce9

Effect of `open.expansion = 2.0` (current situation):

https://github.com/rerun-io/rerun/assets/49431240/9ae1b454-c5ff-4da0-8d3b-d3b030da779f

martenbjork, 2023-07-24:
I think scaling is a disorienting pattern. I think we should remove it across the app. Colors + hover cursor should be enough to signify interactivity. So, I think we should remove scaling from any interactive elements for now.

If we remove it, we will spot where states are missing and then we can fix that upstream in the design system 🙂

abey79, 2023-07-26:
#2796 reverted by #2824

PR #2796 (2023-07-24): Disable expansion effects in the UI
### What

Fixes #2727
Fixes #2726
Fixes #2730

Questions for the reviewers:
- This affects the popup menu controls. Due to their styling the expansion effect was invisible, but resulted in the greyed hover area to be larger by 2px all around. The difference is visible in the videos below. Is that OK? @martenbjork 
- This PR removes the expansion effect from the blueprint tree's triangles, and the play/pause/etc. buttons (desired, not shown in the videos below). Did I miss some other, undesirable side-effect elsewhere?

Before:

https://github.com/rerun-io/re [... cut]
commit 9ca1f3bc (2023-07-24): Disabled expansion effects in the UI (#2796)

### What

Fixes #2727
Fixes #2726
Fixes #2730

Questions for the reviewers:
- This affects the popup menu controls. Due to their styling the
expansion effect was invisible, but resulted in the greyed hover area to
be larger by 2px all around. The difference is visible in the videos
below. Is that OK? @martenbjork
- This PR removes the expansion effect from the blueprint tree's
triangles, and the play/pause/etc. buttons (desired, not shown in the
videos below). Did I miss some other, undesirable side-effect elsewhere?

Before:

https://github.com/re [... cut]
PR #2851 (2023-07-28): Migrate to custom checkbox/radio_value UI
### What

This PR migrates our checkboxes and radio_values to a custom implementation in `re_ui`. Currently, the only change is the removal of the extension effect. In the future, the customisation might go deeper (e.g. custom tick mark, a variant for popup menu, etc.), possibly all the way to copy/pasting and rewriting the entire checkbox code.

This PR also introduces a lint to forbid the use of `ui.checkbox()`/`ui.radio_value()`.

Fixes #2727

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/mai [... cut]
commit 4ce7a405 (2023-07-28): Migrated to custom checkbox/radio_value UI (#2851)

### What

This PR migrates our checkboxes and radio_values to a custom
implementation in `re_ui`. Currently, the only change is the removal of
the extension effect. In the future, the customisation might go deeper
(e.g. custom tick mark, a variant for popup menu, etc.), possibly all
the way to copy/pasting and rewriting the entire checkbox code.

This PR also introduces a lint to forbid the use of
`ui.checkbox()`/`ui.radio_value()`.

Fixes #2727

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/reru [... cut]
closed 2023-07-28 by abey79