rerun-io/rerun #2734: Improve padding in dropdown menus

martenbjork, 2023-07-18:


abey79, 2023-07-27:
@martenbjork The usual "egui" way of doing this is to use checkboxes instead of regular buttons as it's currently done. This is visible, e.g., in the rerun menu:

[image]

What immediately stands out is:
1. the "v" check mark shapes (which comes from egui) is different from that of sigma
2. the default background of the checkbox is ever so slightly different of the menu's background, so we can see it

How do you feel about (1)?  Note that this is the same checkmark for regular, non-menu checkbox:

[image]

As for (2), I'll fix the checkbox color to match the menu, let me know if I should go the other way round.

martenbjork, 2023-07-28:
1. I think having consistent iconography across the UI is pretty important. Using the icons from Figma will ensure that.
2. I find the EGUI way of doing "checkable menu items" not great. As a rule, when hovering an interactive item, the entire item should respond as one big interactive surface. That feels predictable and reliable. In constrast, when you hover one thing and another thing changes visually, the UI feels very finicky and unreliable. (What should I click? The menu item? The checkbox? Will clicking the checkbox close the menu?)

Rerun: 

https://github.com/rerun-io/rerun/assets/875708/bf19e53b-fb45-4142-aadc-20f4957402ba

Mac OS:

https://github.com/rerun-io/rerun/assets/875708/c5cb61be-fc1c-4792-90aa-f09967f79b97

Figma:

https://github.com/rerun-io/rerun/assets/875708/eaf63a23-c5d9-4e90-8bfc-38ce12f03272

martenbjork, 2023-07-28:
An example from Rerun getting this right. Look how big, trustworthy and predictable this toggle feels. Super solid! (Especially when we get rid of the scaling of the arrow, then it will just be one big hover area reacting predictably.)

https://github.com/rerun-io/rerun/assets/875708/04bf929e-780c-40da-8e19-8533f05e1797