rerun-io/rerun #2758: UI point size is wrong when zooming in 2D image

jleibs, 2023-07-19:
Run colmap, in both the 3D and the 2D view set the UI points to 10.
Then Zoom in in the image.

[image]

Wumpf, 2023-07-20:
Seems to be a regression since last release:
Here https://demo.rerun.io/version/latest/examples/structure_from_motion/ it still works and here it doesn't https://demo.rerun.io/pr/2751/examples/structure_from_motion/

Wumpf, 2023-07-24:
I think I got this wrong, on main this seems to work fine. There is the smaller problem of point radius not being a radius but a diameter, but otherwise it seems fine:

Screenshots with 100% scaling, meaning pixel size == point size.
Zoomed in:
[image]
And zoomed out a bit:
[image]

jleibs, 2023-07-24:
This still repros for me on main.  Investigating a bit further, The key appears to be choosing a window aspect ratio such that when zoomed all the way out, the image is scaled based on the *horizontal* dimension.

[image]
[image]

If instead the window is larger than the source image, or the scaling is driven off the vertical dimension, then everything works fine.

Wumpf, 2023-07-25:
excellent find/observation. fix coming

Wumpf, 2023-07-25:
was almost certainly pre-existing

commit 381a847a (2023-07-25): Fix ui-scale based point/line sizes incorrectly scaled when zooming based on horizontal dimension (#2805)

### What

Fixes #2758 
* #2758

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun.io](https://demo.rerun.io/pr/2805) (if
applicable)

- [PR Build Summary](https://build.rerun.io/pr/2805)
- [Docs
preview](https://rerun.io/preview/pr%3Aandre [... cut]