rerun-io/rerun #2767: Data logged from a forked child process does not show up in the viewer.

Victorlouisdg, 2023-07-20:
**Describe the bug**
Data logged from a forked child process does not show up in the viewer. This used to work in rerun version 0.4, but doesn't anymore in 0.7. I'm logging from Python, if I start child processes with the `"fork"` method (default on Unix), no logs come through. However, if I set `multiprocessing.set_start_method("spawn")`, the logging works as expected.

**To Reproduce**
* Install rerun 0.7.
* Run the [python multiprocessing example](https://github.com/rerun-io/rerun/tree/main/examples/python/multiprocessing) from the current main branch (with the `@rr.shutdown_at_exit` line commented out.)

**Expected behavior**
I expected to see logs from 4 processes, as in the screenshot on the example page, however I only see the log from the parent process.

**Screenshots**
[image]

**Desktop:**
 - OS: Ubuntu 20.04

**Rerun version**
rerun_py 0.7.0 [rustc 1.69.0 (84c898d65 2023-04-16), LLVM 15.0.7] x86_64-unknown-linux-gnu prepare-0.7 9cf3033, built 2023-06-16T15:47:32Z

Wumpf, 2023-07-20:
I think this is a duplicate to #1921 which got fixed on latest.
Any chance you could give this a spin with the latest dev-build via `pip install --pre -f https://build.rerun.io/commit/96f5dde/wheels --upgrade rerun-sdk`?

Victorlouisdg, 2023-07-25:
Thanks, I can confirm that solves my issue. After installing that build, all 4 tasks show up in the viewer as expected.

closed 2023-07-25 by Victorlouisdg