rerun-io/rerun #2782: `components::ColorRGBA` should either document how the public `u32` works, or use an array

kpreid, 2023-07-23:
This is a Rust API design nitpick.

As of `rerun 0.7.0`, [`rerun::components::ColorRGBA`](https://docs.rs/rerun/0.7.0/rerun/components/struct.ColorRGBA.html) publicly uses a `u32` to store the color. This is an unnecessary complication, because `u32`s have endianness which confuses the question of which color component ordering in memory is actually meant. (For example, on little-endian architectures — most computers today — the components are actually stored as what someone discussing image formats would call ABGR, not RGBA.) In my opinion, the representation should actually be either `[u32; 4]` (with an explicitly documented meaning of each array index) or a struct

```rust
#[repr(C)]
struct ColorRGBA {
    r: u8,
    g: u8,
    b: u8,
    a: u8,
}
```

Either one of these nails down the memory representation unambiguously, and will be representation-compatible (i.e. transmutable) with someone else's RGBA color type when it's actually RGBA and not ABGR.

Another option is to make the existing `u32` field private, so that the interpretation of it is entirely an implementation detail of `rerun`.

Wumpf, 2023-07-26:
agreed! Either the struct as sugested or `rgba: [u8; 4]` would be an improvement

emilk, 2023-08-08:
I think `[u8; 4]` is the best choice here (especially since Arrow will split a `struct` into separate columns), with nice helper constructors.

emilk, 2023-09-11:
Let's improve the docs for 0.9, but not change the type until 0.10

emilk, 2023-09-12:
I've improved the docs now, but will leave this issue open as a reminder to change the implementation to the more natural `[u8; 4]`