rerun-io/rerun #2785: `Arrow3D` migration to archetypes

teh-cmc, 2023-07-24:
Turn into sub-issues as needed.

- [ ] Python support
  - [ ] Extensions
  - [ ] Unit test (constructors, serialization)
  - [ ] Doc example
- [ ] Rust support
  - [ ] Extensions
  - [ ] Unit test (constructors, rust-to-rust roundtrip)
  - [ ] Doc example
- [ ] C++ support
  - [ ] Extensions
  - [ ] Unit test (constructors, serialization)
  - [ ] Doc example
- [ ] Cross-language roundtrip test
- [ ] Remove old components
  - [ ] Clean up `re_components`
  - [ ] Port viewer to use new types and queries

teh-cmc, 2023-07-28:
Grabbing this one as this overlaps with the migration of `Transform3D` which touches `Vec*D` and `Mat*x*`.

Wumpf, 2023-08-07:
all the infrastructure to move forward is in place!
C++ codegen is missing a few features, but we'll add them as we go

PR #2916 (2023-08-07): C++ codegen extensions, archetype tests, array ctors
### What

Introduces a simple extension system for C++ codegen: Add an extra cpp file that will be compiled as part of the SDK. A section between two markes is copied into the generated hpp as part of generation (typically this section is removed from compilation via #ifdef)

Adds extensions to:
* color
* vec2/vec3/vec4
* quaternion
* origin3d
* point2d
* point3d
* arrow3d

... and uses them to simplify examples and test code!

All fully supported archetypes now have a simple test that checks that the base interface works and that we can ser [... cut]
commit 52d66ea0 (2023-08-07): C++ codegen extensions, archetype tests, array ctors (#2916)

### What

Introduces a simple extension system for C++ codegen: Add an extra cpp
file that will be compiled as part of the SDK. A section between two
markes is copied into the generated hpp as part of generation (typically
this section is removed from compilation via #ifdef)

Adds extensions to:
* color
* vec2/vec3/vec4
* quaternion
* origin3d
* point2d
* point3d
* arrow3d

... and uses them to simplify examples and test code!

All fully supported archetypes now have a simple test that checks that
the base interface works and that w [... cut]
closed 2023-08-07 by Wumpf