rerun-io/rerun #2786: `Rect2D` migration to archetypes

teh-cmc, 2023-07-24:
Turn into sub-issues as needed.

- [ ] Python support
  - [ ] Extensions
  - [ ] Unit test (constructors, serialization)
  - [ ] Doc example
- [ ] Rust support
  - [ ] Extensions
  - [ ] Unit test (constructors, rust-to-rust roundtrip)
  - [ ] Doc example
- [ ] C++ support
  - [ ] Extensions
  - [ ] Unit test (constructors, serialization)
  - [ ] Doc example
- [ ] Cross-language roundtrip test
- [ ] Remove old components
  - [ ] Clean up `re_components`
  - [ ] Port viewer to use new types and queries

emilk, 2023-08-30:
Discussion here: https://rerunio.slack.com/archives/C041NHU952S/p1691680747657179?thread_ts=1691674786.938629&cid=C041NHU952S

Summary: `archetype Box{2,3} = { Point{2,3}, HalfExtents{2,3}, Optional<Rotation{2,3}> }`

emilk, 2023-09-11:
We need this:
* https://github.com/orgs/rerun-io/projects/1/views/11?pane=issue&itemId=38209991

PR #3351 (2023-09-19): Fill in remaining C++ samples and tests that were blocked by lack of Box2D
### What

* Fixes #2787 
* Fixes #2789
* Fixes #2786 
   * once  #3348 is in

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun.io](https://demo.rerun.io/pr/3351) (if applicable)

- [PR Build Summary](https://build.rerun.io/pr/3351)
- [Docs preview](https://rerun.io/prev [... cut]
commit 2004ab77 (2023-09-19): Fill in remaining C++ samples and tests that were blocked by lack of Box2D (#3351)

### What

* Fixes #2787 
* Fixes #2789
* Fixes #2786 
   * once  #3348 is in

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun.io](https://demo.rerun.io/pr/3351) (if
applicable)

- [PR Build Summary](https://build.rerun.io/pr/3351)
- [Docs
preview](https://reru [... cut]
closed 2023-09-19 by Wumpf