rerun-io/rerun #2814: Spurious warnings when clicking space views in the blueprint tree

abey79, 2023-07-25:
**Describe the bug**

In some cases, clicking space views in the blueprint tree generates warnings such as:
```
 egui_tiles::tiles] Failed to find tile #3 during make_active
```

Not all examples are subject to this. At least objectron is (both Python and Rust).

**To Reproduce**
Steps to reproduce the behavior:
1. Run example
2. Click on a space view in the blueprint tree
3. Observe stderr

emilk, 2023-08-25:
If we can't find this quickly, let's just mute the warnings from `egui_tiles` so we don't spam the user

abey79, 2023-08-26:
I bisected that and the culprit is, somewhat unsurprisingly, ec0c7cf02b43fda8a261c48b65b02efeb8ad5a08. So somewhere in #2465 or https://github.com/rerun-io/egui_tiles/pull/7.

closed 2023-09-13 by emilk