rerun-io/rerun #2825: The end of `MsgSender` and streamlining logging APIs across all SDKs

teh-cmc, 2023-07-26:
I've been talking about the retirement of `MsgSender` for a long time but never made an actual issue, apparently.

`MsgSender` is heavy, clunky and almost entirely superfluous now that `RecordingStream` handles all the hard parts of logging for all 3 SDKs.
With the new archetype-based API, it will somehow become even more superfluous than it already is.

The removal of `MsgSender` is also an opportunity to streamline the behavior of all 3 SDKs behind 2 entrypoints:
- `log(entity_path, archetype)` (aliased to `log_archetype()`), which is what I like to call the high-level object-oriented API.
- `log_components(entity_path, components...)`, which is what I like to call the low-level ECS API.

### Example of using the high-level API

As a user, the high-level API gives you guarantees that things will _just work_ :tm:.

You don't need to worry about how things compose together, rather you think in more familiar terms: objects.

In Rust:
```rust
// logging a Points3D archetype, both the timepoint and the recordingstream are picked up from TLS:
rerun::log("points", &Points3D::new([(0.0, 0.0, 0.0), (1.0, 1.0, 1.0)])
    .with_radii([1.0, 2.0])
    .with_colors([0xFF0000]))?;

// which is the same as:
rerun::log_archetype("points", &Points3D::new([(0.0, 0.0, 0.0), (1.0, 1.0, 1.0)])
    .with_radii([1.0, 2.0])
    .with_colors([0xFF0000]))?;

// we can also imagine exposing one with an explicit recordingstream argument:
rerun::log2(&rec_stream, "points", &Points3D::new([(0.0, 0.0, 0.0), (1.0, 1.0, 1.0)])
    .with_radii([1.0, 2.0])
    .with_colors([0xFF0000]))?;
```

And in Python:
```python
# logging a Points3D archetype, both the timepoint and the recordingstream are picked up from TLS:
rr.log("simple", rr.Points3D([[0, 0, 0], [1, 1, 1]], radii=[1, 2], colors=0xFF0000))

# which is the same as:
rr.log_archetype("simple", rr.Points3D([[0, 0, 0], [1, 1, 1]], radii=[1, 2], colors=0xFF0000))
```

And probably something very similar in C++, according to my recent discussions with @Wumpf!

### Example of using the low-level API

As a user, the low-level API gives you access to all of Rerun's power, but comes at a cost: you need to think in terms of composition, and understand all the rules involved (partial updates, joins, instance keys, archetype definitions...).
And of course it's much more verbose and much less friendly (can't autocomplete when we don't know what you're looking for to begin with!).

This allows you to incrementally build up archetypes over multiple timestamps, to do partial updates, to build custom views, to use extension components, etc.

In Rust:
```rust
// logging that same 3D point cloud but using composition this time:
rerun::log_components(
    "points",
    [rrc::Point3D::new(0.0, 0.0, 0.0), rrc::Point3D::new(1.0, 1.0, 1.0)],
    [rrc::Radius::new(0.0), rrc::Radius::new(1.0)],
    [rrc::Color::new(0xFF0000)],
)?;

// or, expressed in a nicer way ():
rerun::log_components(
    "points",
    [(0.0, 0.0, 0.0), (1.0, 1.0, 1.0)].map(rrc::Point3D::from),
    [0.0, 1.0].map(rrc::Radius::from),
    [0xFF0000].map(rrc::Color::from),
)?;
```

Python:
```python
# logging that same 3D point cloud but using composition this time:
rr.log_components(
    "simple",
    rrc.Point3DArray([[0, 0, 0], [1, 1, 1]]),
    rrc.RadiusArray([1, 2]),
    rrc.ColorArray([0xFF0000]),
)
```

And probably something very similar in C++.

Wumpf, 2023-07-26:
Love it and agree with the plan overall!
People surely gonna have opinions on what is alias how to "log", but personally I like the way you're going about it here!

Small'ish improvement suggestion: Rust `rerun::log2(&rec_stream, ..)` looks a bit wonky (I'm always trying to avoid numbering methods). Maybe we could do `rec_stream.log(...)` etc., i.e. have the static/global methods just forward to the global "default" recording stream in a transparent way?

teh-cmc, 2023-07-26:
@jleibs raised an interesting point in IM:
> Yeah, something like this seems quite interesting. My other concern with our current usage of fbs is someone that is using fbs in their own project is going to see our fbs files and be like "oh great, rerun supports fbs, I can just send my fbs-encoded data right into rerun".

teh-cmc, 2023-07-27:
Something I forgot to mention here but that we've talked about a few times with @jleibs before: the high-level {archetype/object}-oriented API is not only easier to use, it also opens many performance opportunities in the backend.

Assuming that the high-level API always logs _the entire set of components for an archetype_ at once (which means logging empty arrays for unspecified fields, as opposed to not logging anything!), then you know that all the data needed for the archetype at a given timestamp is available **in a single datastore row**!
This means you don't need to look for each component individually, and can bypass the component-level join entirely (which means bypassing the entire instance-key machinery too!).
It all comes down to a single `DataStore::get()` + some trivial logic for empty arrays and splats.

That's probably an order of magnitude improvement, if not more, in terms of performance.

How does the backend know that it can take the happy path though? I see two solutions:
- The implicit one: if you know what archetype you're querying for, and know that the row of the primary component holds values for all other components of the archetype, then you know you're on the happy path.
- The explicit path: toggle an extra bit somewhere when the data is logged using the archetype-based APIs.

Wumpf, 2023-07-27:
One thing I'm going a bit back and forth now again is how the `log_components` api is packaged:
Being explicit about this being a different, more advanced operation than just `log` is nice, but maybe also hinders discoverability where using it might be advantegous? I'm thinking of cases where you have a lot of components, let's say initially logged as an archetype, but then change only a single component array repeatedly over time.

To clarify, this comes down to
```python
rr.log("simple", rr.Points3D([[0, 0, 0], [1, 1, 1]], radii=[1, 2], colors=0xFF0000))
rr.log("simple", rrc.ColorArray([0xFF00FF])
```
versus
```python
rr.log("simple", rr.Points3D([[0, 0, 0], [1, 1, 1]], radii=[1, 2], colors=0xFF0000))
rr.log_components("simple", rrc.ColorArray([0xFF00FF])
```
(python, but applies to most languages)

In the past I was leaning very much to the "everything is log", but I do see why you prefer not hide the difference.

jleibs, 2023-07-27:
A very similar version of this I had proposed was instead:
```
rr.log("simple", rr.Points3D([[0, 0, 0], [1, 1, 1]], radii=[1, 2], colors=0xFF0000))
rr.log("simple", rr.Points3D.components(colors=[0xFF00FF])
```

This would still give us the *context* that you are updating a Points3D archetype, and so it wouldn't let you make mistakes by logging components that aren't part of the archetype you expect, but still making you be a bit explicit about the fact that you aren't logging fully specified archetype instances and as such you're doing something a bit advanced.

teh-cmc, 2023-08-25:
We'll follow the plan in:
- https://github.com/rerun-io/rerun/issues/3103

PR #3184 (2023-09-05): Introduce new `RecordingStream` logging APIs (sunset `MsgSender`)
**Commit by commit**

Implements the new logging APIs, which supersede the ol' `MsgSender`:
```rust
pub fn log(
    &self,
    ent_path: impl Into<EntityPath>,
    arch: &impl Archetype,
) -> RecordingStreamResult<()>;

pub fn log_archetype(
    &self,
    ent_path: impl Into<EntityPath>,
    timeless: bool,
    arch: &impl Archetype,
) -> RecordingStreamResult<()>;

pub fn log_component_lists<'a>(
    &self,
    ent_path: impl Into<EntityPath>,
    timeless: bool,
    num_instances: u32,
    comp_lists: impl IntoIterator<Item = [... cut]
commit e1bde648 (2023-09-05): Introduce new `RecordingStream` logging APIs (sunset `MsgSender`) (#3184)

**Commit by commit**

Implements the new logging APIs, which supersede the ol' `MsgSender`:
```rust
pub fn log(
    &self,
    ent_path: impl Into<EntityPath>,
    arch: &impl Archetype,
) -> RecordingStreamResult<()>;

pub fn log_archetype(
    &self,
    ent_path: impl Into<EntityPath>,
    timeless: bool,
    arch: &impl Archetype,
) -> RecordingStreamResult<()>;

pub fn log_component_lists<'a>(
    &self,
    ent_path: impl Into<EntityPath>,
    timeless: bool,
    num_instances: u32,
    comp_lists: impl IntoIterat [... cut]
closed 2023-09-05 by teh-cmc