rerun-io/rerun #2830: Improve user-experience of loading blueprints from older versions

jleibs, 2023-07-26:
At the moment we get a somewhat cryptic error message.

teh-cmc, 2023-07-26:
Adding to that: we currently get the same exact error (something along the lines of "trying to load a `0.7.0` dataset into a `0.8.0` viewer") whether we're loading an outdated _data recording_ or an outdated _blueprint recording_.

So not only the UI might be in a weird state because an outdated blueprint has been loaded, but on top of that the user is frantically searching for the source of outdated data... of which there in fact isn't any.

jleibs, 2023-07-26:
The main complication is this warning is produced long before we've even deserialized the bits that tell us whether it's an RRD or a Blueprint:

https://github.com/rerun-io/rerun/blob/c3487e21f28b778e29a6b2ee2642bb7e8ac2f5eb/crates/re_log_encoding/src/decoder.rs#L69-L91

emilk, 2023-08-25:
Let's see if we can find a quick fix for this, else push it to 0.9.1

emilk, 2023-08-28:
Let's add a `strict` flag to the loading that aborts on incompatible version (instead of warning, then trying to load it and hoping for the best)

closed 2023-09-12 by emilk