rerun-io/rerun #2840: Transform visualization doesn't work for views with content

jleibs, 2023-07-26:
**Describe the bug**
Based on https://github.com/rerun-io/rerun/pull/2577
I was expecting to be able to turn on transform visualization in the api transform_3d demo.

**To Reproduce**
```
python tests/python/test_api/main.py --test transforms_rigid_3d
```

**Expected behavior**
Expect to find an entity property to enable / disable visualization of the transforms, but no such option exists.

Wumpf, 2023-07-27:
They are apparently just not where you expect them?
[image]

[image]

Wumpf, 2023-07-27:
we don't inheritance/global options on this one yet, that's probably what we're missing? I'm a bit worried our systems aren't up for the task there yet - we already have the problem that disabling interactivity on a group correctly propagates but still shows as "enabled" on all children (on an individual basis they are!)

Wumpf, 2023-08-25:
This is one of the many problems related to how we do grouping. We need to rethink that in general which is a different issue than just any particularities on transform controls.

closed 2023-08-25 by Wumpf