rerun-io/rerun #2843: Show hovered Space View position for 2D & 3D space views

Wumpf, 2023-07-27:
For all hover events in 2D & 3D space views we already *know* a position in that space (i.e. a position relative to the space's root entity). We should just display it.

Figure out where to show it in an unobtrusive way (as part of the "on hover stack"?)

(**not** covered by this:
Showing a 3D position for hovering a depth image in 2D. We also have the necessary data (otherwise we couldn't project rays into 3D. But it's even less clear where to show that in a way it's not confusing -- maybe in the respective 3D spaces only, relative to those roots?)

nikolausWest, 2023-07-27:
@martenbjork worth a look

martenbjork, 2023-07-28:
> @martenbjork worth a look

Happy to jam on this whenever we're ready to work on it. I might need a visual explainer :)

jleibs, 2023-07-31:
Showing the *projected* coordinates relative to a different origin would be very powerful here as well.

Specifically I'm imagining hovering in the 2d view of the depth sensor, but seeing a value of an origin such as "world" when we have a pinhole projection linking us to that space.

Wumpf, 2023-09-06:
Related to
* https://github.com/rerun-io/rerun/pull/3227
* https://github.com/rerun-io/rerun/issues/3226