rerun-io/rerun #2857: Increase min width of FPS / 1.0x inputs in timeline

martenbjork, 2023-07-28:


abey79, 2023-07-28:
There are two aspects here I believe:
1) the fact that the widget's width vary between display and edit stage
2) the fact that the widget's width vary based on its content (without even going into edit mode) — this is visible by dragging it to vary its value

Both lead to everything on the right of the widget being interacted with to move around. Is this story about both aspects? If so, we'll need to figure out a min width that cover all cases (more or less), so the highlight might feel less tightly adjusted with the text:
[image]

(this is exaggerated on purpose)

martenbjork, 2023-08-04:
1. Hmm, that's not good. As a rule, no element's size should change between states.
2. Agreed, let's figure out a decent min width for these boxes. It doesn't have to be perfect, we just have to make the jumping around stop 😀

emilk, 2023-08-08:
This can be controlled by `egui::Spacing::interact_size.x`