rerun-io/rerun #2868: Add `RotationMatrix` similar to `Quaternion` and `RotationAxisAngle`

roym899, 2023-07-30:
**Is your feature request related to a problem? Please describe.**
Common way of representing transforms is to have a rotation matrix and a 3D translation. Right now the direct way to log these is via `TranslationAndMat3`. However it would be more natural to use `Rigid3D`. However that only supports `Quaternion` and `RotationAxisAngle` for the rotation argument.

**Describe the solution you'd like**
Add a class `RotationMatrix` and support it in both `Rigid3D` and `TranslationRotationScale3D`. Related to that it might be worth renaming `Quaternion` to `RotationQuaternion` to have consistent naming for the rotation representations, i.e., `RotationMatrix`, `RotationQuaternion`, `RotationAxisAngle`.

Might be nice to have a check on construction that the matrix is actually orthogonal and its determinant is 1.

Wumpf, 2023-07-30:
The reason I decided so far against passing matrices to `Rigid3D` is that it looks like it implies that we'd be doing validation at _some_ point if the matrix passed is actually just a rotation.
My issue with that is that we don't actually _need_ to do that, after all `TranslationAndMat3` just takes about any matrix3.

That said, this very much ties into the problem of allowing to pass in 4x4 matrices for affine transformations. I suppose for both cases we need to do validation internally upon use (that's the easiest place) and simply display a warning once that a given entity violates the implied contract.

If we go through with both changes we end up with this tree of unions for a transform

Transform:
- [ ] *[new]* Mat4 - `Mat4x4`
- [ ] TranslationAndMat3 - `Option<Vec3D>`, `Option<Mat3x3>`
- [ ] Rigid3D - `Option<Vec3D>`, `Option<Scale>`, `Option<Rotation>`
  - `Scale`:
     - [ ] Uniform  - `f32`
     - [ ] NonUniform - `Vec3D`,
  - `Rotation`:
     - [ ] Quaternion - `Quaternion`
     - [ ] AxisAngle - `Vec3D`, `Angle`
         - `Angle`:
              - [ ] Radians - `f32`
              - [ ] Degrees - `f32`
      - [ ] **[new]** Matrix - `Mat3x3`

Does that makes sense?
not sure how I feel about `RotationQuaternion`, need also to see what it's gonna look like in the new API.

roym899, 2023-07-30:
Another option might be to have a Rotation class that can be initialized either from a Quaternion, AxisAngle or Matrix. More similar to what [scipy](https://docs.scipy.org/doc/scipy/reference/generated/scipy.spatial.transform.Rotation.html) does.

Wumpf, 2023-07-31:
> So maybe also keeping the TranslationRotationScale would be more correct in that case?

ah, yeah I got that wrong.

> Instead of two / three separate classes. 

We'll need to see how that works out with our data-definition files and codegen in the near future: `Rotation` as such is a `union` of a bunch of different sub-values. We surely should be abled to make it look like in scipy

Wumpf, 2024-07-22:
fixed in https://github.com/rerun-io/rerun/pull/6929
    * the layout described in there is not happening like that, but I believe in spirit we ticked the most important things:
        * ✔️ `RotationQuat is a thing by now
        * ✔️ `TransformMat3x3` is a thing, not quite the asked `RotationMatrix` but close enough
        * ✔️ a rotation matrix and scale & translate can live side by side

closed 2024-07-22 by Wumpf