rerun-io/rerun #2869: Default view for single image + camera is off / too small default image plane distance

roym899, 2023-07-30:
**Describe the annoyance**
Not sure, how this is set, but in [this rrd file (0.8.0)](https://github.com/rerun-io/rerun/files/12208971/data.zip), a single image and camera is logged, and the resulting default view (i.e., after repopulating) is not good.

[image]

It seems like the default image plane distance (0.025 in this case) is too small in this case, but I guess equivalently the default zoom / and position is wrong. However, currently the easiest way to get back to a good view is to increase the image plane distance.

**Expected behavior**
A default view similar to the one shown below would be nice.

[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

Wumpf, 2023-07-30:
The problem in that scene is that there it first logs a tensor and _then_ a camera. Despite being on the same timestamp on the _iteration_ timeline, the order on the log time affects the outcome of the "accumulated bounding box" which is used to determine the default camera position: The picture is put into 3D space, resulting in a huge bounding box and _then_ gets scaled down by means of being part of a camera projection.
[image]

The simplest way out of this - and the logical consequence of the 3D/2D space view split - is that we simply say that a 3D space can't display a 2D image unless we have some description of the spatial relationship as given here with a pinhole camera. After all, the 2D image in the first frame on the log time line doesn't logically make sense (it's huge ;)).
By solving this deeper issue, the bounding box and resulting camera failure described in this ticket should resolve itself.

@nikolausWest I _think_ this solution is in the spirit of your push to 2D/3D space separation but I wanted to double check - is it? 😄

nikolausWest, 2023-07-31:
> The simplest way out of this - and the logical consequence of the 3D/2D space view split - is that we simply say that a 3D space can't display a 2D image unless we have some description of the spatial relationship as given here with a pinhole camera.

@Wumpf - this makes sense to me. It also brings up that we should have some easy way to place 2D images into 3D through a more direct method than a pinhole projection

teh-cmc, 2023-10-12:
- Fixed by #3822

teh-cmc, 2023-10-12:
- Fixed by #3822

closed 2023-10-12 by teh-cmc