rerun-io/rerun #2870: Support row vector convention when logging rotation matrix

roym899, 2023-07-30:
When logging transformation matrices, rerun currently assumes that they are multiplied from the left hand side to a vector represented as a column (let's call that column vector convention). As discussed, for example, [here](https://www.scratchapixel.com/lessons/mathematics-physics-for-computer-graphics/geometry/row-major-vs-column-major-vector.html) another (less common) convention is to right multiply. One notable library that uses such a row vector convention is [Pytorch3D](https://pytorch3d.readthedocs.io/en/latest/modules/transforms.html#pytorch3d.transforms.Transform3d). 

When given a row convention rotation matrix, it has to be transposed before logging to rerun. It would be better if this could be specified as an argument. 

**Additional context**
Related to, but to but not the same as #2189. Also related to #2868.

Wumpf, 2023-07-30:
It's not written down yet, but internally we decided that we don't want _storage representation_ to be configurable. Row/column majorness is treated just like endianess, meaning it's not something we allow specifying in the data store. This is opposed to _meaning_ of data which is why we for example don't want to normalize radians or degrees - a conversion would change user intent and introduce an error unlike arrangement of bytes (like row/column major or endianess).

That said, the API should accomodate for Pytorch and make it easy to get the data in. Also, the viewer should make it very clear what it shows and maybe even make it configurable globally how it shows matrices.

roym899, 2023-07-30:
I think this is a separate issue from row-major / column-major storage representation. Rather this is about whether a rotation matrix is left-multiplied or right-multiplied to a vector and mathematically related to that whether a 3D vector is interpreted as a 3x1 matrix or 1x3 matrix. 

I *think* some libraries choose the latter because it allows expressing transforms of many points without requiring a transpose. I.e., you have an Mx3 matrix of M points, you can transform these points with a simple (M,3) x (3,3) matrix multiplication. In that case however the (3,3) rotation matrix is transposed from its usual format.

Wumpf, 2023-07-31:
The only place where we actually leak out multiplication order is the "combined" versions of transform, e.g. `TranslationRotationScale` where there needs to be a logical order of what is applied when. But otherwise it shouldn't play much of a rule on our api if we work with column or row vectors, i.e. chain from the left or the right?

roym899, 2023-07-31:
Implicitly its leaked out into the rotation matrix as well. The same rotation is represented by the transposed rotation matrix when using row vectors / multiplying from the right.

nikolausWest, 2023-08-07:
> I think some libraries choose the latter because it allows expressing transforms of many points without requiring a transpose. I.e., you have an Mx3 matrix of M points, you can transform these [...]

I think this implies that we should also allow for some ergonomic way to log points that are (M, 3) arrays instead of the (3, M) that we take now

roym899, 2023-08-07:
Current way to log is actually (M,3), which seems more common than (3,M), even when the code transforms from the left. But I agree, specifying the meaning of each dimension when logging points would be nice. An argument taking "BC" or "CB" (to specify batch dimension and channel dimension) would be nice to have, and is more readable compared to transposing prior to logging.
I think that's nearly the same issue as #2817 is for images.