rerun-io/rerun #2876: Expose a way for the SDK to notify that it is loading/computing stuff?

teh-cmc, 2023-07-31:
At the moment, the only way for client code to indicate that it's currently busy loading stuff is to dump info into the terminal (e.g. `tqdm` progress bars).

It's not a great experience though, it's easy to miss and there's nothing in the viewer itself indicating that anything's happening.

I'm wondering if we shouldn't provide a simple API that the SDK can call to indicate in the UI that stuff is happening, please wait.

nikolausWest, 2023-08-07:
Just to clarify, are you're mainly concerned with the impression a user gets when they run an example that is doing heavy processing? I think we can get pretty far here by:
- Indicating in the UI when we're connected to a data source (SDK)
- Updating our slower examples to log text like: "downloading data", "computing X" so that something happens in the UI by default

teh-cmc, 2023-08-07:
> Just to clarify, are you're mainly concerned with the impression a user gets when they run an example that is doing heavy processing?

Yes. I myself often wonder if any progress is being made, even though I've run these examples hundreds of times...

>     * Indicating in the UI when we're connected to a data source (SDK)
> 
>     * Updating our slower examples to log text like: "downloading data", "computing X" so that something happens in the UI by default

Sounds like a great first step; and a pattern that users can then apply to their own programs. :+1: