rerun-io/rerun #2880: Unexpected egui_tiles behaviour with space views

abey79, 2023-07-31:
It's currently very easy to trigger unexpected behaviours related to egui_tiles when moving around space views. I identified two cases:

1) "de-tabbing": sometimes, tabbed views (which are in most cases the result of user actions) are de-tabbed when a new space view is created
2) "grid with empty column": the grid layout doesn't suppress an empty column.

I also observed a varient of (2) where the grid layout ended up being:

<table>
  <tbody>
    <tr>
      <td>something</td>
      <td>something</td>
<td>something</td>
    </tr>
    <tr>
      <td>something</td>
      <td><em> empty </em></td>
      <td><em> empty </em></td>
    </tr>
  </tbody>
</table>

Here is a video showing both (1) and (2):

https://github.com/rerun-io/rerun/assets/49431240/63a1c7e9-8e66-42c9-963c-b01728964518

emilk, 2023-09-12:
* https://github.com/rerun-io/egui_tiles/pull/26 fixes nr 2

emilk, 2023-09-12:
Problem nr 1 comes from `has_been_user_edited` not being set when a user drags around tiles using `egui_tiles`.

closed 2023-09-13 by emilk