rerun-io/rerun #2889: Crash on shutdown when using thread local recording stream, Mac only (?)

Wumpf, 2023-08-01:
Does not repro on Linux, observed so far only on my mac.

When doing this from C++
```cpp
rr::RecordingStream stream("test", kind);
stream.set_thread_local();
```
I get a crash on application shutdown with the following callstack:
```
thread '<unnamed>' panicked at 'use of std::thread::current() is not possible after the thread's local data has been destroyed', library/std/src/thread/mod.rs:742:35
stack backtrace:
   0:        0x1030a0ff0 - std::backtrace_rs::backtrace::libunwind::trace::h2000fb4d08dbcc59
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/../../backtrace/src/backtrace/libunwind.rs:93:5
   1:        0x1030a0ff0 - std::backtrace_rs::backtrace::trace_unsynchronized::h2b5e61495350674d
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2:        0x1030a0ff0 - std::sys_common::backtrace::_print_fmt::h05f5bfbdb3415936
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/sys_common/backtrace.rs:65:5
   3:        0x1030a0ff0 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h105074e3d85f800b
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/sys_common/backtrace.rs:44:22
   4:        0x1030eb0f4 - core::fmt::write::h34766cf8fff7af1e
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/core/src/fmt/mod.rs:1232:17
   5:        0x103096d68 - std::io::Write::write_fmt::hd64c4cf6e7adea59
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/io/mod.rs:1684:15
   6:        0x1030a0e04 - std::sys_common::backtrace::_print::hd92783a665d3ebfb
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/sys_common/backtrace.rs:47:5
   7:        0x1030a0e04 - std::sys_common::backtrace::print::h2a6828a537036cf9
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/sys_common/backtrace.rs:34:9
   8:        0x1030a388c - std::panicking::default_hook::{{closure}}::h4e82ce6ccef941b2
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/panicking.rs:271:22
   9:        0x1030a35e4 - std::panicking::default_hook::h29f62f8795c5cb00
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/panicking.rs:290:9
  10:        0x1030a3e48 - std::panicking::rust_panic_with_hook::h19862cbd0fbda7ba
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/panicking.rs:692:13
  11:        0x1030a3d7c - std::panicking::begin_panic_handler::{{closure}}::h3f3626935e1669fe
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/panicking.rs:583:13
  12:        0x1030a1410 - std::sys_common::backtrace::__rust_end_short_backtrace::h5054ef52bd507d0a
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/sys_common/backtrace.rs:150:18
  13:        0x1030a3ad8 - rust_begin_unwind
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/panicking.rs:579:5
  14:        0x103123928 - core::panicking::panic_fmt::h7e47e10600a90221
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/core/src/panicking.rs:64:14
  15:        0x1030e7df4 - core::panicking::panic_display::h940323af6be747f0
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/core/src/panicking.rs:147:5
  16:        0x1030e7db8 - core::panicking::panic_str::h7ca06b5260522b55
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/core/src/panicking.rs:131:5
  17:        0x1031238f4 - core::option::expect_failed::h3fd9ac3cfb0d19fd
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/core/src/option.rs:2045:5
  18:        0x10308b070 - core::option::Option<T>::expect::hfa5d7210f95a43b5
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/core/src/option.rs:905:21
  19:        0x10308b070 - std::thread::current::h46f377b22b1f0c11
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/thread/mod.rs:742:35
  20:        0x1031129ac - crossbeam_channel::context::Context::new::h3fbb24b343b5b8b9
                               at /Users/andreas/.cargo/registry/src/github.com-1ecc6299db9ec823/crossbeam-channel-0.5.8/src/context.rs:75:25
  21:        0x102d181d8 - crossbeam_channel::context::Context::with::{{closure}}::h6ddd9cafffc8aec1
                               at /Users/andreas/.cargo/registry/src/github.com-1ecc6299db9ec823/crossbeam-channel-0.5.8/src/context.rs:65:36
  22:        0x102d181d8 - core::result::Result<T,E>::unwrap_or_else::h4047661a446f8fcc
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/core/src/result.rs:1465:23
  23:        0x102d181d8 - crossbeam_channel::context::Context::with::haf7c05038ad35a20
                               at /Users/andreas/.cargo/registry/src/github.com-1ecc6299db9ec823/crossbeam-channel-0.5.8/src/context.rs:55:9
  24:        0x102d181d8 - crossbeam_channel::flavors::zero::Channel<T>::recv::h1a2572704f92b4ca
                               at /Users/andreas/.cargo/registry/src/github.com-1ecc6299db9ec823/crossbeam-channel-0.5.8/src/flavors/zero.rs:310:9
  25:        0x102d3b25c - crossbeam_channel::channel::Receiver<T>::recv::h13fe8fa2bede526e
                               at /Users/andreas/.cargo/registry/src/github.com-1ecc6299db9ec823/crossbeam-channel-0.5.8/src/channel.rs:816:43
  26:        0x102d64058 - re_log_types::data_table_batcher::DataTableBatcherInner::flush_blocking::h85690880bd26535b
                               at /Users/andreas/dev/rerun-io/rerun/crates/re_log_types/src/data_table_batcher.rs:356:9
  27:        0x102c2f988 - re_log_types::data_table_batcher::DataTableBatcher::flush_blocking::hc669352c402b1960
                               at /Users/andreas/dev/rerun-io/rerun/crates/re_log_types/src/data_table_batcher.rs:326:9
  28:        0x102c2f988 - <re_sdk::recording_stream::RecordingStreamInner as core::ops::drop::Drop>::drop::h3fc93ff5f922ba89
                               at /Users/andreas/dev/rerun-io/rerun/crates/re_sdk/src/recording_stream.rs:431:9
  29:        0x102c2e354 - core::ptr::drop_in_place<re_sdk::recording_stream::RecordingStreamInner>::h1fa43b3772060a69
[... 14 lines cut ...]
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/core/src/ptr/mod.rs:490:1
  37:        0x102c21de4 - core::ptr::drop_in_place<core::cell::RefCell<core::option::Option<re_sdk::recording_stream::RecordingStream>>>::hbacb929c3a90db0d
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/core/src/ptr/mod.rs:490:1
  38:        0x102c21de4 - core::ptr::drop_in_place<core::option::Option<core::cell::RefCell<core::option::Option<re_sdk::recording_stream::RecordingStream>>>>::h4910982148057a66
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/core/src/ptr/mod.rs:490:1
  39:        0x102c21de4 - core::mem::drop::hf69534690d0b17fb
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/core/src/mem/mod.rs:979:24
  40:        0x102c21de4 - std::thread::local::fast::destroy_value::{{closure}}::h15ca396f2a6d1b28
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/thread/local.rs:1036:13
  41:        0x102c21de4 - core::ops::function::FnOnce::call_once::h260e1e35f76a094f
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/core/src/ops/function.rs:250:5
  42:        0x102c21de4 - <core::panic::unwind_safe::AssertUnwindSafe<F> as core::ops::function::FnOnce<()>>::call_once::h1fcb162acfce0f6c
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/core/src/panic/unwind_safe.rs:271:9
  43:        0x102c21de4 - std::panicking::try::do_call::hbabad351ad3f2b64
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/panicking.rs:487:40
[... cut ...]
  46:        0x102c21de4 - std::thread::local::fast::destroy_value::h1778796e8865515c
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/thread/local.rs:1033:25
  47:        0x1030ac950 - std::sys::unix::thread_local_dtor::register_dtor::run_dtors::hfed6ec2d63bd81ed
                               at /rustc/84c898d65adf2f39a5a98507f1fe0ce10a2b8dbc/library/std/src/sys/unix/thread_local_dtor.rs:79:17
./rerun_cpp/build_and_run_tests.sh: line 16: 16757 Abort trap: 6           ./build/rerun_cpp/tests/rerun_sdk_tests
```

emilk, 2023-08-08:
Feels likely this could be reproduced from the Rust SDK too, no?

emilk, 2023-10-19:
```rs
fn main() {
    let stream = rerun::RecordingStreamBuilder::new("rerun_example_app_test")
        .buffered()
        .unwrap();
    rerun::RecordingStream::set_thread_local(rerun::StoreKind::Recording, Some(stream));
}
```

also crashes:

```
thread '<unnamed>' panicked at 'use of std::thread::current() is not possible after the thread's local data has been destroyed', library/std/src/thread/mod.rs:711:35
stack backtrace:
   0: rust_begin_unwind
             at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library/std/src/panicking.rs:593:5
   1: core::panicking::panic_fmt
             at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library/core/src/panicking.rs:67:14
   2: core::panicking::panic_display
             at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library/core/src/panicking.rs:150:5
   3: core::panicking::panic_str
             at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library/core/src/panicking.rs:134:5
   4: core::option::expect_failed
             at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library/core/src/option.rs:1952:5
   5: core::option::Option<T>::expect
             at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library/core/src/option.rs:898:21
   6: std::thread::current
             at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library/std/src/thread/mod.rs:711:35
```

emilk, 2023-10-19:
`RUST_BACKTRACE=1 cargo test -p re_sdk --lib` also fails on Mac, with:

```
thread panicked while processing panic. aborting.
error: test failed, to rerun pass `-p re_sdk --lib`

Caused by:
  process didn't exit successfully: `/Users/emilk/code/rerun/rerun/target/debug/deps/re_sdk-a7f79267aeda2288` (signal: 5, SIGTRAP: trace/breakpoint trap)
```

emilk, 2023-10-19:
The problem is this: when the thread dies, the destructor of the thread-local `RecordingStream` is called, which in turn uses `crossbeam` channels. `crossbeam` calls `std::thread::current` which apparently is not allowed in this context.

I think this is likely a bug in the thread-teardown code of `std` - destructors of thread-locals should be called before tearing down `std::thread:current`.

emilk, 2023-10-19:
Minimal reproduce:

```rs
use std::cell::RefCell;

struct Foo {}

impl Drop for Foo {
    fn drop(&mut self) {
        println!("Dropping Foo");
        println!("Thread: {:?}", std::thread::current().id()); // panics
    }
}

thread_local! {
    static THREAD_LOCAL: RefCell<Option<Foo>> = RefCell::new(None);
}

fn main() {
    THREAD_LOCAL.with(|cell| *cell.borrow_mut() = Some(Foo {}));
    println!("So far so good!");
}
```

emilk, 2023-10-19:
* https://github.com/rust-lang/rust/issues/116933

closed 2023-10-20 by emilk