rerun-io/rerun #2905: Make rerun_c smaller

Wumpf, 2023-08-03:
`rerun_c` depends on `re_types` through `re_sdk`All it needs is like `Data{Cell/Row}` and `RecordingStream`. Maybe we should have an option an `re_sdk` to discard all of the archetype stuff.

_Originally posted by @teh-cmc in https://github.com/rerun-io/rerun/pull/2901#discussion_r1282992216_

emilk, 2023-08-08:
Note: cargo features should be additive, and `--all-features` should enable the maximum set, hence we should not have features to _disable_ stuff, instead we should have an opt-out (default) feature for all the standard datatypes/components/archetypes.

emilk, 2023-10-24:
This should be fairly easy now - I think it only needs to depend on `re_types_core`

Wumpf, 2023-11-02:
apparently there's no easy way to change linker settings per crate. Guess this part is less important anyways since we're not shipping a shared library but just a static one which can be stripped during _actual_ linking (static libraries are typically just packaged up object files with minimal to no linking)

PR #4147 (2023-11-07): Reduce rerun_c library size (by depending on fewer unnecessary crates)
### What

* Fix #2905 
   * reduces Rerun C size from 41.0MiB to 34.2MiB (MacOS aarch64)
   * feels like there's more we should be able to do, but this PR stops the bleed of having it become larger as we add new types!
   * note that since this is a static library, typically the actual stripping occurs when linking the final binary, so while it's great that our library gets smaller for many reasons, it's not _critical_
* `re_log_types` no longer depends on `re_types`
* `re_sdk` no longer depends on `re_memory`
* `re_sdk` no [... cut]
commit 316bdddd (2023-11-07): Reduce rerun_c library size (by depending on fewer unnecessary crates) (#4147)

### What

* Fix #2905 
   * reduces Rerun C size from 41.0MiB to 34.2MiB (MacOS aarch64)
* feels like there's more we should be able to do, but this PR stops the
bleed of having it become larger as we add new types!
* note that since this is a static library, typically the actual
stripping occurs when linking the final binary, so while it's great that
our library gets smaller for many reasons, it's not _critical_
* `re_log_types` no longer depends on `re_types`
* `re_sdk` no longer depends on `re_memory`
* `re_sdk` [... cut]
closed 2023-11-07 by Wumpf