rerun-io/rerun #2917: [C/C++] Introduce unified error handling

Wumpf, 2023-08-05:
* Related to #2873 

Right now the C api crashes on most errors.
C++ has a log library dependency but so far we ended up not using it.

Suggesting to introduce an error value enum or struct (to be able to convey more complex error state, i.e. another tagged union, but from C?) and drop our logging library dependency instead.

emilk, 2023-08-10:
for the C boundary we can do:

```C
typedef uint32_t ErrorCode;

struct Error {
     ErrorCode error_code;
     char description[512];
};

void rr_foo(…, Error* error);
```

That will report errors from Rust to C.

emilk, 2023-08-10:
For our initial setup function (`connect`, `save`, etc) we should return a `Result`, similar to `arrow::Result`. We should use the same `ErrorCode` `enum` as in the C API.

The `Result` class should have nice helpers for logging warnings to stderr or throwing it as an exception.

jleibs, 2023-08-10:
Still disagree on this being called a `warning_handler`.

In my opinion, passing in data that does not conform to the schema is an error. "The application or library couldn't complete an operation."  The application could not encode the passed data into a *valid* encoding according to the specified schema.
Specifically, this is a user-error. The only way this happens is if the user wrote incorrect code that called our APIs. This is always a bug they should fix in their code.

The path we are on is specifically this one:
 -> "Sometimes an Err is handled by logging it as WARNING and then running some fallback code."

We are allowing the user to choose whether to handle this Error by logging a warning, and continuing with the fallback of using a weaker best-effort schema, or by treating it as an error and taking more aggressive action. But the thing they are handling is still an error.

closed 2023-08-17 by Wumpf