rerun-io/rerun #2918: Failed to send to Rerun server at 127.0.0.1:9876: Broken pipe (os error 32) , with the example for `structure from motion`

dinusha94, 2023-08-07:
**Describe the bug**
When I try to run the example for `structure from motion` I am getting the following warnings and there is no visualization.
```
Reading sparse COLMAP reconstruction
Building visualization by logging to Rerun
[2023-08-07T03:35:15Z WARN  re_sdk_comms::buffered_client] Failed to send message after 3 attempts: Failed to send to Rerun server at 127.0.0.1:9876: Broken pipe (os error 32)
[2023-08-07T03:35:18Z WARN  re_sdk_comms::buffered_client] Dropping messages because tcp client has timed out.
[2023-08-07T03:35:18Z WARN  re_sdk_comms::buffered_client] Dropping messages because tcp client has timed out.
[2023-08-07T03:35:21Z WARN  re_sdk_comms::tcp_client] Tried to flush while TCP stream was still Pending. Data was possibly dropped.
[2023-08-07T03:35:21Z WARN  re_sdk_comms::tcp_client] Tried to flush while TCP stream was still Pending. Data was possibly dropped.

```

**To Reproduce**

1. Steps to reproduce the behavior:
2. pip install --upgrade rerun-sdk  # install the latest Rerun SDK
3. git clone git@github.com:rerun-io/rerun.git  # Clone the repository
4. cd rerun
5. git checkout latest
6. pip install -r examples/python/structure_from_motion/requirements.txt
7. python examples/python/structure_from_motion/main.py

**Expected behavior**
visualization of 3d points

**Backtrace**
```
[2023-08-07T03:35:10Z INFO  winit::platform_impl::platform::x11::window] Guessed window scale factor: 1
[2023-08-07T03:35:10Z WARN  wgpu_core::instance] Missing downlevel flags: DownlevelFlags(SURFACE_VIEW_FORMATS)
    The underlying API or device in use does not support enough features to be a fully compliant implementation of WebGPU. A subset of the features can still be used. If you are running this program on native and not in a browser and wish to limit the features you use to the supported subset, call Adapter::downlevel_properties or Device::downlevel_properties to get a listing of the features the current platform supports.
WARNING: lavapipe is not a conformant vulkan implementation, testing use only.

thread 'ThreadId(1)' panicked at 'The given device doesn't support the required downlevel capabilities for the given hardware caps DeviceCaps { tier: FullWebGpuSupport, max_texture_dimension2d: 16384 }.
            Required:
            DownlevelCapabilities { flags: DownlevelFlags(COMPUTE_SHADERS | FRAGMENT_WRITABLE_STORAGE | INDIRECT_EXECUTION | BASE_VERTEX | READ_ONLY_DEPTH_STENCIL | NON_POWER_OF_TWO_MIPMAPPED_TEXTURES | CUBE_ARRAY_TEXTURES | COMPARISON_SAMPLERS | INDEPENDENT_BLEND | VERTEX_STORAGE | ANISOTROPIC_FILTERING | FRAGMENT_STORAGE | MULTISAMPLED_SHADING | DEPTH_TEXTURE_AND_BUFFER_COPIES | WEBGPU_TEXTURE_FORMAT_SUPPORT | BUFFER_BINDINGS_NOT_16_BYTE_ALIGNED | UNRESTRICTED_INDEX_BUFFER | FULL_DRAW_INDEX_UINT32 | DEPTH_BIAS_CLAMP | VIEW_FORMATS | UNRESTRICTED_EXTERNAL_TEXTURE_COPIES | SURFACE_VIEW_FORMATS), limits: DownlevelLimits, shader_model: Sm4 }
            Actual:
            DownlevelCapabilities { flags: DownlevelFlags(COMPUTE_SHADERS | FRAGMENT_WRITABLE_STORAGE | INDIRECT_EXECUTION | BASE_VERTEX | READ_ONLY_DEPTH_STENCIL | NON_POWER_OF_TWO_MIPMAPPED_TEXTURES | CUBE_ARRAY_TEXTURES | COMPARISON_SAMPLERS | INDEPENDENT_BLEND | VERTEX_STORAGE | FRAGMENT_STORAGE | MULTISAMPLED_SHADING | DEPTH_TEXTURE_AND_BUFFER_COPIES | WEBGPU_TEXTURE_FORMAT_SUPPORT | BUFFER_BINDINGS_NOT_16_BYTE_ALIGNED | UNRESTRICTED_INDEX_BUFFER | FULL_DRAW_INDEX_UINT32 | DEPTH_BIAS_CLAMP | VIEW_FORMATS | UNRESTRICTED_EXTERNAL_TEXTURE_COPIES), limits: DownlevelLimits, shader_model: Sm5 }', re_renderer/src/context.rs:142

   0: backtrace::backtrace::trace
   1: backtrace::capture::Backtrace::new

Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues
Reading sparse COLMAP reconstruction
Building visualization by logging to Rerun
[2023-08-07T03:35:15Z WARN  re_sdk_comms::buffered_client] Failed to send message after 3 attempts: Failed to send to Rerun server at 127.0.0.1:9876: Broken pipe (os error 32)
[2023-08-07T03:35:18Z WARN  re_sdk_comms::buffered_client] Dropping messages because tcp client has timed out.
[2023-08-07T03:35:18Z WARN  re_sdk_comms::buffered_client] Dropping messages because tcp client has timed out.
[2023-08-07T03:35:21Z WARN  re_sdk_comms::tcp_client] Tried to flush while TCP stream was still Pending. Data was possibly dropped.
[2023-08-07T03:35:21Z WARN  re_sdk_comms::tcp_client] Tried to flush while TCP stream was still Pending. Data was possibly dropped.

```

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

**Rerun version**
rerun_py 0.8.0 [rustc 1.69.0 (84c898d65 2023-04-16), LLVM 15.0.7] x86_64-unknown-linux-gnu main df05f49, built 2023-07-27T17:36:23Z

Any help will be highly appreciated
Thank you

Wumpf, 2023-08-07:
Thanks for filing a report @dinusha94 ! It seems you're running with a software rasterizer (lavapipe) which isn't fully supported by Rerun. You might need to look into installing a graphics driver on your system. Are you running this from a VM?
You can also try working around it by forcing the GL backend instead of Vulkan (which we're defaulting to on Linux) by starting rerun with `WGPU_BACKEND=gl`, i.e. `WGPU_BACKEND=gl python examples/python/structure_from_motion/main.py` in your case.

Any of this working for you?

dinushazoomi, 2023-08-07:
After running this `WGPU_BACKEND=gl python examples/python/structure_from_motion/main.py` I am getting the following massage. I do not have much knowledge of the `GL backend`,  is this massage saying I don't have it installed?

```
[2023-08-07T11:01:13Z INFO  winit::platform_impl::platform::x11::window] Guessed window scale factor: 1
libEGL warning: DRI2: failed to authenticate

thread 'ThreadId(1)' panicked at 'Error in Surface::configure: Validation Error

Caused by:
    Requested usage is not supported
', wgpu-0.16.1/src/backend/direct.rs:734

   0: backtrace::backtrace::trace
   1: backtrace::capture::Backtrace::new

Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues
Reading sparse COLMAP reconstruction
Building visualization by logging to Rerun
[2023-08-07T11:01:18Z WARN  re_sdk_comms::buffered_client] Failed to send message after 3 attempts: Failed to send to Rerun server at 127.0.0.1:9876: Broken pipe (os error 32)
[2023-08-07T11:01:21Z WARN  re_sdk_comms::buffered_client] Dropping messages because tcp client has timed out.
[2023-08-07T11:01:21Z WARN  re_sdk_comms::buffered_client] Dropping messages because tcp client has timed out.
[2023-08-07T11:01:24Z WARN  re_sdk_comms::tcp_client] Tried to flush while TCP stream was still Pending. Data was possibly dropped.
[2023-08-07T11:01:24Z WARN  re_sdk_comms::tcp_client] Tried to flush while TCP stream was still Pending. Data was possibly dropped.
```

dinushazoomi, 2023-08-07:
@Wumpf I managed to run it with my Windows machine, can we use `.mvs` file for visualizing

Wumpf, 2023-08-08:
nice! Hoping this isn't too inconvenient for you!
Keeping this issue open for tracking lack of lavapipe/vm support

dinushazoomi, 2023-08-09:
Hi @Wumpf, can I use this to display data from OpenMVS

Wumpf, 2023-08-09:
Sure, I don't seem why not, should be a good fit. We don't have a direct integration though yet, so you'd need to extract the data into Numpy arrays and then pass it to Rerun's sdk from there

Wumpf, 2023-08-23:
gathering all crashes of this kind now on #3089 and will take it from there (first getting a good error message, then potential follow-up for actual support)