rerun-io/rerun #2919: C++ SDK V1

Wumpf, 2023-08-07:
Tracking issue for everything we want to have for a first version of an officially supported C++ SDK.

General guideline:
* feature complete: Log everything that the *new* object oriented Python API can log
* works on Linux & Mac, Windows optional at this point
* _basic_ distributability - we have an installation guide but it doesn't need to me smooth at this point

TODO, roughly ordered by priority:
* [ ] #2917
* [ ] feature complete logging #2795
* [ ] add (at least one) high level example
* [ ] getting started doc page (implying a rough idea on how to distribute)
* [ ] improve header quality - #2873
* [ ] #2903

Nice to have for V1
* [ ] generate documentation and link from website
* [ ] example for custom component
* [ ] check on distribution size - #2905
* [ ] easy way to start Rerun from C++ (like `spawn` in Python)

Future (i.e. excluded from this release)
* [ ] support one or more C++ package manager
* [ ] C++14 support (requiring C++17 so far)

emilk, 2023-08-08:
It seems like we should add API docs (generated from docstrings) to this as well

Wumpf, 2023-08-08:
that's covered by `generate documentation and link from website`

closed 2023-10-30 by emilk