rerun-io/rerun #2924: `face_tracking` example fails with new `AnnotationContext` implementation

abey79, 2023-08-07:
**Describe the bug**

It looks like `rr.ClassDescription()` now requires and argument that wasn't formerly required.

```
❯ python examples/python/face_tracking/main.py 
INFO: Created TensorFlow Lite XNNPACK delegate for CPU.
Traceback (most recent call last):
  File "/Users/hhip/src/rerun/examples/python/face_tracking/main.py", line 428, in <module>
    main()
  File "/Users/hhip/src/rerun/examples/python/face_tracking/main.py", line 422, in main
    run_from_video_capture(args.camera, args.max_dim, args.max_frame, args.num_faces)
  File "/Users/hhip/src/rerun/examples/python/face_tracking/main.py", line 311, in run_from_video_capture
    detector = FaceDetectorLogger(video_mode=True)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/hhip/src/rerun/examples/python/face_tracking/main.py", line 119, in __init__
    rr.ClassDescription(keypoint_connections=[(0, 1), (1, 2), (2, 0), (2, 3), (0, 4), (1, 5)]),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/hhip/src/rerun/rerun_py/rerun_sdk/rerun/_rerun2/datatypes/class_description.py", line 53, in __init__
    classdescription_init(self, *args, **kwargs)
TypeError: classdescription_init() missing 1 required positional argument: 'info'
```

abey79, 2023-08-07:
This example is likely broken as well: https://www.rerun.io/docs/reference/data_types/annotation_context#connections-example

emilk, 2023-08-17:
I believe this was introduced in https://github.com/rerun-io/rerun/pull/2895/files#diff-e4f8d37313c7297c2faac15acce4fa15501d41993f79b717e817213f8c20764eR30 

`struct ClassDescription` has a `pub info: crate::datatypes::AnnotationInfo,` which before used the `AnnotationInfo::default()`:

```rust
#[derive(Clone, Debug, Default, Eq, PartialEq)]
pub struct AnnotationInfo {
    /// `ClassId` or `KeypointId` to which this annotation info belongs.
    pub id: u16,

    /// The label that will be shown in the UI.
    pub label: Option<crate::datatypes::Label>,

    /// The color that will be applied to the annotated entity.
    pub color: Option<crate::datatypes::Color>,
}
```

closed 2023-08-17 by emilk

emilk, 2023-08-17:
This is not a problem on 0.8.0, so no need for a fix for 0.8.1