rerun-io/rerun #2963: Log file:line of rerun log call

emilk, 2023-08-11:
When user logs some data with Rerun we should log the `file:line` of that log call.

A user could use this to figure out where a log call came from. For instance, they may see an image in Rerun and wonder where in their source code that was logged.

When a user misuses Rerun we could also tell the user which file:line in their code cause the problem.

### How to find the file:line
* C++: use a `RERUN_LOG` macro that reads `__FILE__` and `__LINE__`
* Python: there is certainly some magic for this
* Rust: Use `#[track_caller]` and `Location::caller()`

### How to store it
We could have a `CallLocation` component with the `file:line` in it.
To save store space we could consider hashing that, and only logging the hash for each log call (except the first one).

emilk, 2023-10-31:
For C++20 there is [`std::source_location`](https://en.cppreference.com/w/cpp/utility/source_location/current), and there might be compiler extensions for earlier C++ versions.

Wumpf, 2025-03-10:
cc: @SeaOtocinclus 
this is the thing we talked about on a call just now, looks like Emil already wrote down pretty much all of it quite a while ago 😅