rerun-io/rerun #3008: Mechanism to filter and/or expire data from the view

jleibs, 2023-08-16:
A frequent user-request is to stop displaying data which has been logged but is no longer valid or relevant to the view.

The only current work-around for this is to clear the data manually using the clearing APIs (https://ref.rerun.io/docs/python/0.8.0/common/clearing_entities/), or equivalently logging empty batches.

However, tracking and manually expiring data can be both complicated and error-prone, especially if it originates from within a stateless function. The application itself may not even have access to the state to know what data to expire.

An alternative way to handle this would be to introduce some form of "TTL" associated with the data. Ideally this could either be logged with the data directly, or overridden from the blueprint configuration.

A few complications to keep in mind:
 - Data is often logged against multiple timelines. Does a TTL need to be specified for every timeline?
 - How does this interact with visible history?
 - If users want to specify data is only valid "at the time logged" (i.e. TTL = 0), how does this interact with the timeline selection? If all data is logged synchronously but against a continous timeline we potentially need some version of "snap to valid timepoint" on the time controls.
 - Should TTL be per-entity? per-component? per-instance? 
   - Single mono-value per entity is probably a good starting point until we have demonstrated need of more fine-grained control.
 - Performance-wise should we handle this inside the query-engine (more performant), or should space-view-parts still receive the instances and then filter them internally (easier / more flexible)?

Qubi0-0, 2025-04-03:
Just bumping this! Would really like to see this fix too!

Such a hassle now to remove entities on next timesamp

MarkChenYutian, 2025-12-24:
**Cross-posted at #6136**, just found that this issue seems more active : )

Hi @jleibs , I would like to bump this feature request up for practical usage in the SLAM system visualization!

Currently at time t I'm only visualizing the trajectory from t-1 to t to avoid the O(n) per-frame network bandwidth. To visualize the whole trajectory, I used the time range query to accumulate all the segments to visualize the full trajectory.

However, when a bundle adjustment occured at time x (i.e. some historical trajectory was updated), this method cannot visualize the update properly. 

1. If I log the full updated trajectory at x - then the time-range query will retrieve all the trajectories before and after update, causing annoying duplications in visualizer.

2. The `rr.Clear` mechanism does not work well since we are using time-range query.

3. I can create another entity instance (e.g. if the original path is `/path/trajectory/v0` I can create `/path/trajectory/v1` etc.) and log the new trajectories there. But this will require user to manually turn off the visualization of out-dated entity paths.

This problem can be solved if `rr.Clear` semantic follows the second semantics:

```
t=0 [##]
t=1 [####]
t=2 <clear>
t=3 [###]
t=4 [#]

Range(0, 4) = [(t=3 [###]), (t=4 [#])]
```

I understand that there might have cases where this is not the best rr.Clear semantic, will it be possible to add a boolean flag / argument on rr.Clear to choose between semantic 1 and 2? 

Also, for this very specific use case I won't need the recursive clear (flat-clear is sufficient), as I read from #7482 that recursive `rr.Clear` with time range query is very hard and inefficient to implement.

Thanks!