rerun-io/rerun #3010: Rerun visualizer crashes shortly after start with

pd-nisse, 2023-08-16:
**Describe the bug**
We have a user on a Windows OS that sees `rerun==0.8.0` crash shortly after start. It successfully loads the world map, but after a few seconds just crashes with the following trace:

```
thread 'ThreadId(1)' panicked at 'called `Result::unwrap()` on an `Err` value: Overflow', arrow2-0.17.1/src/array/growable/binary.rs:73

   0: wgpu_render_pass_execute_bundles
   1: wgpu_render_pass_execute_bundles
   2: PyInit_rerun_bindings
   3: PyInit_rerun_bindings
   4: wgpu_render_pass_execute_bundles
   5: wgpu_render_pass_execute_bundles
   6: wgpu_render_pass_execute_bundles
   7: wgpu_render_pass_execute_bundles
   8: wgpu_render_pass_execute_bundles
   9: wgpu_render_pass_execute_bundles
  10: wgpu_render_pass_execute_bundles
  11: wgpu_render_pass_execute_bundles
  12: wgpu_render_pass_execute_bundles
  13: wgpu_render_pass_execute_bundles
  14: wgpu_render_pass_execute_bundles
  15: PyInit_rerun_bindings
  16: PyInit_rerun_bindings
  17: PyInit_rerun_bindings
  18: PyInit_rerun_bindings
  19: PyInit_rerun_bindings
  20: PyInit_rerun_bindings
  21: PyInit_rerun_bindings
  22: PyInit_rerun_bindings
  23: PyInit_rerun_bindings
  24: PyInit_rerun_bindings
  25: PyInit_rerun_bindings
  26: PyInit_rerun_bindings
  27: PyInit_rerun_bindings
  28: DispatchMessageW
  29: CallWindowProcW
  30: wglSwapBuffers
  31: DispatchMessageW
  32: DispatchMessageW
  33: GetClassLongW
  34: KiUserCallbackDispatcher
  35: NtUserDispatchMessage
  36: DispatchMessageW
  37: PyInit_rerun_bindings
  38: PyInit_rerun_bindings
  39: PyInit_rerun_bindings
  40: PyInit_rerun_bindings
  41: PyInit_rerun_bindings
  42: PyInit_rerun_bindings
  43: wgpu_render_bundle_insert_debug_marker
  44: wgpu_render_bundle_insert_debug_marker
  45: PyInit_rerun_bindings
  46: wgpu_render_bundle_insert_debug_marker
  47: wgpu_render_bundle_insert_debug_marker
  48: wgpu_render_bundle_insert_debug_marker
  49: wgpu_render_bundle_insert_debug_marker
  50: PyObject_Str
  51: PyEval_EvalFrameDefault
  52: PyEval_EvalFrameDefault
  53: PyFunction_Vectorcall
  54: PyEval_EvalCodeWithName
  55: PyEval_EvalCodeEx
  56: PyEval_EvalCode
  57: PyCodec_DecodeText
  58: PyCodec_DecodeText
  59: PyEval_EvalFrameDefault
  60: PyFunction_Vectorcall
  61: PyEval_EvalFrameDefault
  62: PyFunction_Vectorcall
  63: PyFunction_Vectorcall
  64: PyVectorcall_Call
  65: PyObject_Call
  66: Py_MakePendingCalls
  67: Py_RunMain
  68: Py_RunMain
  69: Py_Main
  70: <unknown>
  71: BaseThreadInitThunk
  72: RtlUserThreadStart

Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues
```

Unfortunately, I can not narrow it down further because I currently can not repro it on my own system. I was hoping maybe this trace could help anyone from the `rerun` team to have a suspicion for us to dig deeper in. 🙏 

**Desktop (please complete the following information):**
 - OS: Windows 11, python 3.9.13

**Rerun version**
`0.8.0`

**Additional context**
N/A

jleibs, 2023-08-22:
Another user reported that this ocurred shortly after attempting to save a .blueprint file.  You can see the filename location by enabling `RUST_LOG=debug`

There's a good chance the log history will end up looking like:
```
[2023-08-16T14:25:03Z DEBUG re_viewer::store_hub] Saving blueprint for XYZ to "/home/user/.local/share/rerun/blueprints/xyz.blueprint"

thread 'ThreadId(1)' panicked at 'called `Result::unwrap()` on an `Err` value: Overflow', arrow2-0.17.1/src/array/growable/binary.rs:73

<Unhelpful stacktrace follows>
```

Reported work-around is to remove the `.blueprint` file from the folder.  However, if someone that is experiencing this issue is willing to share the blueprint file with rerun it would be very beneficial to help root-causing the problem.

pd-nisse, 2023-08-23:
I can confirm that removing the existing `.blueprint` file resolved the issue. `rerun` works after that and created a new file instead.

jleibs, 2023-08-24:
The two coupled causes of this are:
 - https://github.com/rerun-io/rerun/issues/3097
 - https://github.com/rerun-io/rerun/issues/3098

nikolausWest, 2023-09-01:
Closing now as fixes are in place that significantly mitigate the problem

closed 2023-09-01 by nikolausWest