rerun-io/rerun #3058: setting light mode

edulecom, 2023-08-21:
Hello,

I'm playing with .\rerun\crates\re_ui\examples\re_ui_example.rs and I'm trying to change the theme to light.
I have added only one line in the fn update:

        egui::TopBottomPanel::bottom("bottom_panel")
            .frame(self.re_ui.bottom_panel_frame())
            .show_animated(egui_ctx, self.bottom_panel, |ui| {
                ui.strong("Bottom panel");
                egui::widgets::global_dark_light_mode_buttons(ui); // this line is added
            });  

Effect is strange: only central panel applies the theme. It can be seen in an attached video.
Any help would very appreciated.

https://github.com/rerun-io/rerun/assets/104624770/cf9a8e50-f6f1-4a7d-be5d-a5ce285be531

Wumpf, 2023-08-22:
The out of the box theme settings are bound to not work properly in rerun as we sadly change a lot of the theming colors in Rerun on-the-fly. At least I suspect that's what's most of the issue here.

That said, (re)adding a light mode is something we want to do eventually.
Are you just trying out things or is a Rerun light theme something you'd like to have as well?

edulecom, 2023-08-23:
Thank you for your response. I'm currently in the process of developing an app using re_ui, and while the absence of a light mode isn't a critical issue, having that feature would certainly be a welcome addition.

emilk, 2025-05-14:
Most of the work has landed in https://github.com/rerun-io/rerun/pull/9953 but there are a few things left to do:

* [ ] Fix table headers (e.g. catalogs)
* [ ] Remove the experimental warning
* [ ] Switch the default theme to follow the system
* [ ] Add light mode snapshot tests

lubidkdh, 2025-05-25:
Hello, I'm using rerun through the python language. I hope to make the rerun theme default to light in the form of code. I also hope that the rerun theme can be saved as part of the blueprint in an rbl file. Or is there any other convenient way to adjust the rerun theme? Also, could you specify the address of the usage document for rerun 0.24.0a1 version? I'm very sorry that I haven't found the address of the document for the time being. Thank you very much.

lubidkdh, 2025-05-26:
Ok. Thank you very much for your response!

Wumpf, 2025-06-13:
We have experimental light made in the upcoming 0.24.
@emilk / @gavrelina how far are we with it - are we going to get it out of experimental for 0.24 or are we even already done here? :)

PR #10500 (2025-07-03): Stabilize light mode
### Related
* Closes https://github.com/rerun-io/rerun/issues/10289
* Closes #3058 

### What
This is one of the last remaining blockers for light mode!

This affects any part of the UI where we use blue to indicate "selection", including:
* `ListItem`:s (e.g. blueprint panel tree)
* Selection panel breadcrumbs
* Selected tabs in viewport
* Play-button

The difference is largest in light mode.

[image]
[image]
[image]
[image]

### Other

In addition, I also made sure the background of the `>` collapsing icon gets filled when hovered:

[image]

### Other other
This also ena [... cut]
commit 40109aa6 (2025-07-03): Stabilize light mode (#10500)

### Related
* Closes https://github.com/rerun-io/rerun/issues/10289
* Closes #3058 

### What
This is one of the last remaining blockers for light mode!

This affects any part of the UI where we use blue to indicate
"selection", including:
* `ListItem`:s (e.g. blueprint panel tree)
* Selection panel breadcrumbs
* Selected tabs in viewport
* Play-button

The difference is largest in light mode.

[image]

[image]

[image]

[image]

### Other

In addition, I also made sure the background of the `>` collapsing icon
gets filled when hovered:

[image]

### Other other
 [... cut]