rerun-io/rerun #3062: Segmentation fault when dragging view

apbr, 2023-08-22:
**Description**
I get a segmentation fault when dragging the view:

> [2023-08-22T08:34:26Z INFO  re_sdk_comms::server] Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.
> [2023-08-22T08:34:28Z INFO  re_sdk_comms::server] New SDK client connected: 127.0.0.1:51564
> 
> Rerun caught a signal: SIGSEGV
> Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
> Report bugs: https://github.com/rerun-io/rerun/issues
> 
>    1: re_crash_handler::install_signal_handler::signal_handler
>    2: <unknown>
>    3: _ZN4llvm7CmpInst6CreateENS_11Instruction8OtherOpsENS0_9PredicateEPNS_5ValueES5_RKNS_5TwineEPNS_10BasicBlockE
> 
> 
> Rerun caught a signal: SIGSEGV
> Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
> Report bugs: https://github.com/rerun-io/rerun/issues
> 
> Segmentation fault (core dumped)

**Reproduce**
1. Run the following code snippet:
```Rust
use rerun::{
    components::{Point3D, Radius},
    MsgSender, RecordingStreamBuilder,
};
use anyhow::Result;

fn main() -> Result<()> {
    let recording = RecordingStreamBuilder::new("Events")
        .connect(rerun::default_server_addr(), None)?;
    let v: Vec<Point3D> = Vec::new();
    MsgSender::new("points")
        .with_component(&v)?
        .with_splat(Radius(0.001))?
        .send(&recording)?;
    Ok(())
}
```
2. Run rerun with `rerun`.
3. Wait until rerun connected and shows the colored background.
4. Click and drag the colored background
5. Rerun crashes with SIGSEGV

**Expected behavior**
Rerun should not crash

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04.1

**Rerun version**
rerun-cli 0.8.1 [rustc 1.71.0 (8ede3aae2 2023-07-12), LLVM 16.0.5] x86_64-unknown-linux-gnu, built 2023-08-22T08:13:01Z

Wumpf, 2023-08-22:
> _ZN4llvm7CmpInst6CreateENS_11Instruction8OtherOpsENS0_9PredicateEPNS_5ValueES5_RKNS_5TwineEPNS_10B

I'm fairly sure this indicates a crash in llvmpipe, a software implementation of Vulkan. Can you try running with OpenGL by running with `WGPU_BACKEND=gl`? Generally, we don't support software rasterizers at this point, you should install a graphics driver.

Wumpf, 2023-08-22:
(likely related to #2918)

apbr, 2023-08-23:
Yes seems pretty similar to #2918.
I forgot to write that but I'm running Ubuntu 22.04 in a VirtualBox VM. This might explain the issue.
> Can you try running with OpenGL by running with `WGPU_BACKEND=gl`?

This fails like it does in #2918:
```
$ WGPU_BACKEND=gl rerun
[2023-08-23T07:18:41Z INFO  re_sdk_comms::server] Hosting a SDK server over TCP at 0.0.0.0:9876. Connect with the Rerun logging SDK.
[2023-08-23T07:18:41Z WARN  wgpu_hal::gles::egl] 	EGL context: -robust access
[2023-08-23T07:18:41Z WARN  wgpu_hal::gles::egl] Re-initializing Gles context due to Wayland window
[2023-08-23T07:18:41Z WARN  wgpu_hal::gles::egl] 	EGL context: -robust access

thread 'main' panicked at 'Error in Surface::configure: Validation Error

Caused by:
    Requested usage is not supported
', wgpu-0.16.3/src/backend/direct.rs:734

   6: core::panicking::panic_fmt
             at core/src/panicking.rs:67:14
   7: wgpu::backend::direct::Context::handle_error_fatal
   8: <wgpu::backend::direct::Context as wgpu::context::Context>::surface_configure
   9: <T as wgpu::context::DynContext>::surface_configure
  10: wgpu::Surface::configure
  11: egui_wgpu::winit::Painter::configure_surface
  12: egui_wgpu::winit::Painter::resize_and_generate_depth_texture_view_and_msaa_view
  13: pollster::block_on
  14: <eframe::native::run::wgpu_integration::WgpuWinitApp as eframe::native::run::WinitApp>::on_event
  15: eframe::native::run::run_and_return::{{closure}}
  16: winit::platform_impl::platform::wayland::event_loop::EventLoop<T>::run_return
  17: eframe::native::run::with_event_loop
  18: eframe::native::run::wgpu_integration::run_wgpu
  19: eframe::run_native

Troubleshooting Rerun: https://www.rerun.io/docs/getting-started/troubleshooting
Report bugs: https://github.com/rerun-io/rerun/issues
```

`WINIT_UNIX_BACKEND=x11 rerun` Also does not help. (Same crash as just running `rerun`)

Wumpf, 2023-08-23:
gathering all crashes of this kind now on #3089 and will take it from there (first getting a good error message, then potential follow-up for actual support)