rerun-io/rerun #3072: Scrollbar showing up for image that is resized to fit

roym899, 2023-08-22:
**Describe the bug**
Sometimes scrollbars show up for images that actually resize to fit the view (i.e., images that are fully zoomed out). Once they show up you have to increase the size of the view a bit more to get rid of them again.

**To Reproduce**
Not sure exactly when this happens. See video. Seems to happen especially when fiddling around about where the image perfectly fits the view.

https://github.com/rerun-io/rerun/assets/9785832/f1ffb329-7c3a-45eb-b386-1e54f1871ddb

**Expected behavior**
Scrollbars should never show up when they aren't required.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 
**Rerun version**
```
rerun_py 0.8.0 [rustc 1.69.0 (84c898d65 2023-04-16), LLVM 15.0.7] x86_64-unknown-linux-gnu main df05f49, built 2023-07-27T17:36:23Z

https://github.com/rerun-io/rerun/assets/9785832/f1ffb329-7c3a-45eb-b386-1e54f1871ddb

```

Wumpf, 2023-08-25:
scrollbars should be removed all-together, don't think we wrote that down anywhere else

emilk, 2023-08-25:
Will be solved by:
* https://github.com/rerun-io/rerun/issues/2531