rerun-io/rerun #3091: Significant growth in RRD size when loading and then re-saving.

jleibs, 2023-08-24:
**Describe the bug**
Creating an rrd via `--save` produces a file that is significantly smaller than what is produced via "file -> save" in the viewer. This includes opening an existing RRD file and then re-saving it.

**To Reproduce**
Steps to reproduce the behavior:
1. Create an RRD file:
```
$ python examples/python/structure_from_motion/main.py --save direct_save.rrd

$ ls -lh direct_save.rrd
-rw-rw-r-- 1 jleibs jleibs 243M Aug 24 12:44 direct_save.rrd
```

2 . Open the RRD file view the viewer
```
$ rerun direct_save.rrd
```

3. Save the RRD file again via file menu and check the size
```
$ ls -lh direct_save_resave.rrd 
-rw-rw-r-- 1 jleibs jleibs 724M Aug 24 12:45 direct_save_resave.rrd
```

**Expected behavior**
The new RRD file should be around the same size or potentially smaller.

**Desktop (please complete the following information):**
 - Linux: broken
 - Windows: untested
 - OSX: untested

**Rerun version**
- 0.8.1

teh-cmc, 2023-08-25:
This is obviously not optimal (to say the least) but it does align with my expectations considering how it all works today.

The `save` feature in the viewer is currently implemented the lazy way: it turns each individual store bucket into a `LogMsg` and dumps that. So if you have three timelines, you'll end up with roughly three times the data!

It's pretty nice in that saving takes very little extra memory, but we need to throw a table batcher in there!

teh-cmc, 2024-07-10:
Fixed in new ChunkStore.

closed 2024-07-10 by teh-cmc