rerun-io/rerun #3103: Archetypes should implement as_components

nikolausWest, 2023-08-25:
To make it more ergonomic for users to log their own types that are translated into Rerun archetypes, we should introduce a protocol that all archetypes implement:

```python
from typing import Iterable, Protocol

class ComponentsGroup(Protocol):  # TODO: find a better name than ComponentsGroup

    def as_components(self) -> Iterable[rerun.Component]:
        ...
```

We should then make sure that
```python
rr.log("path", SomeArchetype(a, b=b, c=c))
# is equivalent to (and even implemented by)
rr.log_components("path", SomeArchetype(a, b=b, c=c).as_components())
```
That way, if a user already has a type `MyPointCloud` in their code, they can log it as
```python
point_cloud = MyPointCloud(...)
rr.log("path", point_cloud)
```
By letting `MyPointCloud` implement `as_components()`. Could in the simplest case be
```python
class MyPointCloud:
    ...
    def as_components(self) -> Iterable[rr.Component]:
        rertun rr.Points3D(self.positions, ...).as_components()
```

Wumpf, 2023-08-25:
we should do the same in C++, this actually could make `RecordingStream`'s implementation a bit simpler and more transparent to what it does

nikolausWest, 2023-08-25:
In Python, the type alias `Loggable` should be removed and replaced with whatever `ComponentsGroup` ends up being called.

closed 2023-09-19 by jleibs