rerun-io/rerun #3106: Closing the viewer takes several seconds

asuper0, 2023-08-25:
Depends on the size of cached data, closing viewer could take 5 or more than 10 seconds.
Why does it takes so long time? Could you just close it in a rude way?

emilk, 2023-08-25:
I'm guessing it is because of save the app state or blueprint to disk.

You should be able to see what is slow [by starting the profiler](https://github.com/rerun-io/rerun/issues/3083).

* Maybe related to https://github.com/rerun-io/rerun/issues/3098

asuper0, 2023-08-29:
It seems help by resetting the viewer.

The profiler don't show the reason. I tried some times.

[image]

[image]

As the two figures show, the profiler only catch 200~300ms, but the closing time are 700ms for figure 1, and 3000ms for figure 2.

emilk, 2023-09-11:
This will almost certainly be fixed in 0.9 thanks to:

* https://github.com/rerun-io/rerun/issues/3156
* https://github.com/rerun-io/rerun/issues/3098

closed 2023-09-11 by emilk