rerun-io/rerun #3107: Optimize out unnecessary joins across autogenerated instance keys

teh-cmc, 2023-08-25:
The store will auto-generate monotically increasing instance keys when they are not passed in by the user. This allows `re_query`'s join logic to "just work" no matter what but it also means that in most cases we're computing a costly join across 2 components of the same size, both with autogenerated instance keys, even though we already know the answer to that join.

teh-cmc, 2023-08-25:
My proposed solution here would be to completely drop the notion of autogenerated instance keys altogether (which would fix a bunch of other issues).

When we get two components to join and none of them have keys, then we know we can blindly pair them.

teh-cmc, 2023-08-26:
The solution proposed above induces _way_ too many cascading changes resulting from having to make cluster keys optional (internally), since everything has been built on the assumption that they should always be present (and rightly so).

A less invasive solution would be to indicate in the query results whether or not the returned instance keys were autogenerated (which we know since all autogenerated keys of a given length point to the same arrow buffer).

emilk, 2023-09-20:
I think this is high-priority for 0.10, since we have introduces potential performance regressions in 0.9 because of it (https://github.com/rerun-io/rerun/pull/3363)

PR #3377 (2023-09-20): Optimize out unnecessary joins when querying archetypes
What the title says; see the commits for details, it's pretty trivial.

Going from a ~62ms space view build time to ~40ms in the OPF example, so pretty nice gains overall.

OPF, before:
[image]

OPF, after:
[image]

* Fixes #3107

### What

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun [... cut]
commit 84581b76 (2023-09-20): Optimize out unnecessary joins when querying archetypes (#3377)

What the title says; see the commits for details, it's pretty trivial.

Going from a ~62ms space view build time to ~40ms in the OPF example, so
pretty nice gains overall.

OPF, before:
![image
(17)](https://github.com/rerun-io/rerun/assets/2910679/700fe397-7fa2-4f4c-b661-07b9dfb30937)

OPF, after:

[image]

* Fixes #3107
closed 2023-09-20 by teh-cmc