rerun-io/rerun #3152: Expanding streams view doesn't play nicely with other parts of the UI

nikolausWest, 2023-08-30:
See video

https://github.com/rerun-io/rerun/assets/2624717/d48ade66-53aa-44cf-ae2f-a3737769e6f6

How to reproduce:
```bash
pip install --pre -f https://build.rerun.io/commit/d7de7f9/wheels --upgrade rerun-sdk.  # Latest build as of Aug 30th 09.20 CET
python -m rerun_demo
```

abey79, 2023-08-30:
Tagging with 0.9 in case this is a quick fix.

emilk, 2023-08-30:
Let's add a max size to it that leaves at least 25% or 200 pixels level on the screen, or something like that.

nikolausWest, 2023-08-30:
> Let's add a max size to it that leaves at least 25% or 200 pixels level on the screen, or something like that.

If we can avoid a max size that would be nice. It's actually really useful to show just the streams view to explain what's going on in the underlying data model

abey79, 2023-08-31:
Looking into it, it introduce some complexity in quite a few widgets. My proposal:

1) Implement @emilk's suggestion of enforcing a limit at 200px or something.
2) Build (or use if it exists already) a "snapping" resizing system for panel: resizing close to max height should "snap" to using the full height (ie. in that case we dont even draw the viewport/side panels/etc.). Likewise, it would be nice that resizing the time panel all the way down would "snap" to its collapsed state (which would have the added benefit of freeing the top-right button to actually _hide_ the timeline, which can't be done currently).

I'll open a PR for (1) and an issue for (2).

PR #3169 (2023-09-01): Clamp time panel height to avoid visual glitches
### What

Fixes #3152

*Note to reviewer*: I had to resort to some semi-hack to measure the height of "everything above the viewport", i.e. windows title bar vs. heading banner on web (which are significantly different). That's the `screen_header_height` variable. Open to suggestion to make that cleaner.

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] ~~I've included a scre [... cut]
commit 4718642f (2023-09-01): Clamp time panel height to avoid visual glitches (#3169)

### What

Fixes #3152

*Note to reviewer*: I had to resort to some semi-hack to measure the
height of "everything above the viewport", i.e. windows title bar vs.
heading banner on web (which are significantly different). That's the
`screen_header_height` variable. Open to suggestion to make that
cleaner.

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] ~~I've includ [... cut]
closed 2023-09-01 by emilk