rerun-io/rerun #3154: The hover previews on events in the Streams View flicker and block

nikolausWest, 2023-08-30:
See video

https://github.com/rerun-io/rerun/assets/2624717/50c276c9-d783-4478-a2f8-a1b78ac56e0f

As you hover events in the streams view, the preview data UI flickers in and out and covers way too much of the UI. THe effect is quite jarring

How to reproduce:
```bash
pip install --pre -f https://build.rerun.io/commit/d7de7f9/wheels --upgrade rerun-sdk.  # Latest build as of Aug 30th 09.20 CET
python -m rerun_demo
```

emilk, 2023-08-30:
Let's make sure we use `UiVerbosity::Reduced` for these, and if we are, make sure `UiVerbosity::Reduced` is smaller for this example

nikolausWest, 2023-09-01:
> It will not fix the flickering though, which is due to the mouse moving over a bunch of datapoints which are all separated by 2-3 pixels. I'm leaning towards "works as intended".

Once the oversized tooltips are fixed, it might alleviate the problem. If it doesn't, we could consider waiting a little before showing the tooltip. At the same time that will affect the feeling of snappiness. Feels like @martenbjork kind of territory. Either way, happy to hold off on solving that for now if #3171 has a big enough effect

abey79, 2023-09-01:
Sounds good. I'm not sure about the actual psychological impact, but #3171 definitely has reduced tooltip size rather drastically.

martenbjork, 2023-09-01:
Shouldn't the tooltip work harder to get out of the way? Right now, it often covers the cursor, even though there is more space above the cursor where it could go. Here's my reference implementation: https://7e94j1.csb.app/ What do you think about this behavior?

[image]

closed 2023-09-01 by abey79

abey79, 2023-09-01:
I opened #3187 to cover the tooltip behaviour.

PR #3171 (2023-09-01): Limit the size of component tooltips with `UiVerbosity::Reduced`
### What

Limit the size of component tooltips with `UiVerbosity::Reduced`

Fixes #3154

[image]

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun.io](https://demo.rerun.io/pr/3171) (if applicable)

- [PR Build Summary](https://build.rerun.io/pr/3171)
- [Docs preview](https://rer [... cut]
commit 2f016708 (2023-09-01): Limit the size of component tooltips with `UiVerbosity::Reduced` (#3171)

### What

Limit the size of component tooltips with `UiVerbosity::Reduced`

Fixes #3154

<img width="227" alt="image"
src="https://github.com/rerun-io/rerun/assets/49431240/b0bde6b0-52ba-4d9b-9f4f-e709638de983">

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested [demo.rerun.io](https [... cut]